fritzing/fritzing-parts #14: Is there a way to update the parts in Linux Fritzing 0.9.1b

KeithSloan, 2015-03-14:
Is there a way to update the parts in Linux Fritzing 0.9.1b, or is the only way to build from source.
The reason I ask is that I cannot open fzp file extensions in Linux Fritzing 0.9.1b it seems to want them as fzpz. I tried ziping the fzp and renaming to fzpz but that did not work,

el-j, 2015-03-15:
the easiest way could be to clone the parts-branch and overwrite:

	./Fritzing/parts/

solved? ;)

el-j, 2015-03-15:
git clone https://github.com/fritzing/fritzing-parts.git

closed 2015-03-15 by el-j

el-j, 2015-03-15:
and delete the fzp sitting in

	./fritzing/pdb/core
	./fritzing/pdb/obsolete

otherwise there is a problem of doubled module Ids.

KeithSloan, 2015-03-15:
Well I think I followed the steps and I now have various Raspberry Pi's in my parts/core directory, but when I run the Gui I just see Raspberry Pi B+ under Core as was the case before.

el-j, 2015-03-15:
you have to copy the RPi B+ into the sketch,
than use the "variant"-switcher to access  the other versions. 

[image]