fritzing/fritzing-parts #18: ATMega 328p TQFP package has wrong pin name

Elijahg, 2015-06-16:
Pin 14 on the TQFP ATMega328 package has the wrong pin name, it's PB2 (SS/OC1B/PCINT3), but should be PB2 (SS/OC1B/PCINT2).

el-j, 2015-08-08:
it would be nice if you can point to the exact file. i cannot confirm this bug. the tqfp atmega i see hasn't any of the "pcint" descriptions.

thanks

Elijahg, 2015-08-08:
I'm not sure what the filename is, but I've attached screenshots of the exact part. Mousing over the pins shows the incorrect tooltip.
[image]
[image]

closed 2015-11-01 by el-j