fritzing/fritzing-parts #19: Errors in svggen.py

glixx, 2015-07-03:
```diff
File "part-gen-scripts/obsolete/fp2fzfp/svggen.py", line 24
    elif tag == "pad:
                    ^
SyntaxError: EOL while scanning string literal
Sorry: IndentationError: expected an indented block (svggen.py, line 64)
```

aknoerig, 2015-07-10:
This script is no longer in use, hence in the "obsolete" folder. The files it generates are not compatible with current versions of Fritzing, the code only serves as a reference.

closed 2015-07-10 by aknoerig

glixx, 2015-07-10:
It should be fixed or deleted: package cannot be build.

aknoerig, 2015-07-17:
How are you building the app? Maybe it's even better to fix the build script to exclude the obsolete folder.

glixx, 2015-07-17:
If to build RPM package and include fritzing-parts in parts, then error will be.
Will you move all obsoletes soft into separate directory and don't include in release archive for download?