fritzing/fritzing-parts #37: Cannot find the Softpot Membrane Potentiometer part in Fritzing?

aknoerig, 2015-12-10:
_From @eljacko on June 6, 2015 16:25_

Here is a link to evidence it has been used in Fritzing projects:

https://learn.sparkfun.com/tutorials/sik-experiment-guide-for-arduino---v32/experiment-10-reading-a-soft-potentiometer

Please tell me how I can find the Softpot Membrane Potentiometer part.

_Copied from original issue: fritzing/fritzing-app#3092_

karlalhdz, 2017-10-16:
The softpot membrane potentiometer part, its here: https://github.com/sparkfun/Fritzing_Parts/blob/master/products/08680_softpot_membrane_potentiometer_50mm.fzpz

failiz, 2022-11-09:
this can be closed as the part is already in the core

KjellMorgenstern, 2022-11-09:
I tried to search for "Softpot" within the Fritzing Application, but could not find it. Are you sure it is already in core?

I downloaded the part from github.com/sparkfun , and opened / drag and drop it into Fritzing and use it without problems.
The question is, should we sync that repo into Fritzing? There are many sparkfun parts in Fritzing, so I tend to say yes.