fritzing/fritzing-parts #38: Raspberry Pi 2 Pin numbers incorrect

aknoerig, 2015-12-10:
_From @chrisyokum on June 4, 2015 7:12_

The Raspberry Pi 2 header pin numbers are incorrect. Take a look at http://pi.gadgetoid.com/pinout. In the Fritzing the top-left pin (furthest from the Ethernet port) is listed as GND/pin40. On the official documentation it's GND/Pin2. It's confusing when trying to hook things up..

_Copied from original issue: fritzing/fritzing-app#3091_

chrisyokum, 2015-12-10:
The labels are also titled GIPO not GPIO.

vanepp, 2017-12-22:
Actually all of the PI parts are a mess, more internally than external. I have done a corrected version of the Zero (which has the pcb header the wrong size, maybe 2mm but not .1 in as it should be) and I changed the schematic around so that it matches breadboard and posted it in the forums. The issue that I asked about and don't have an answer to is what is the accepted way of making a change that will break existing parts? If I remember correctly the PIs are one of the parts that don't start at pin 0 and more importantly don't have contiguous pin numbers (which breaks the hover over the pin to display the labels function). I can (and I think on the Zero did)) fix all of that, but it will break all existing sketches that use these parts and I don't know how that is dealt with (being relatively new to Fritzing). If you can tell me what you need I can fix up the parts and maybe figure out how to put them in to a pull request (I'm just learning git as well ...) in between fixing the problems in the part check script.

Peter