fritzing/fritzing-parts #40: Raspberry PI Model B Rev 01 - WIN32 causes fritzing Version 0.9.1 to freeze

aknoerig, 2015-12-10:
_From @mikecjudge on December 18, 2014 10:10_

In breadboard view when changing from the default Raspberry PI B+ to the PI B Rev 01 Fritzing Version 0.9.1 (b8d2d5970658f0bed09c661c9ea9a5) becomes unresponsive and freezes.

It works if the Raspberry PI variant is changed in schematic view and then change back to breadboard view.

_Copied from original issue: fritzing/fritzing-app#3035_

aknoerig, 2015-12-10:
_From @el-j on February 19, 2015 16:2_

sorry but i cannot rebuild this behavior. switching between the raspi variants works perfect for me.
what OS do you use? you have the same behavior with other parts? 
what machine do you use?

el-j, 2015-12-11:
since no one have this behavior with this part. i mark this issue as closed

closed 2015-12-11 by el-j