fritzing/fritzing-parts #42: Missing DGS N-Channel Mosfet?

chrisspen, 2016-02-04:
I'm trying to design a circuit with a [ZVN3306A](http://www.redrok.com/MOSFET_ZVN3306A_60V_270mA_5O_Vth2.4_TO-92.pdf) n-channel mosfet, which has a DGS pinout.

However, when I search Fritzing for "n-channel" or "mosfet" all the mosfets only support GSD pinout. I tried switching the "variant" and "package" options, but they're all identical. Is DGS supported?

aknoerig, 2016-02-14:
It's definitely supported, we just didn't make one yet. You're welcome to contribute one. :)

closed 2016-02-14 by aknoerig

chrisspen, 2016-02-25:
What do you mean "it's definitely supported"? If there's no part...

chrisspen, 2016-02-25:
How do I re-order the lead numbers in the part editor? It only appears to let me edit the lead names.

chrisspen, 2016-02-25:
When I click a connector, the "terminal point" panel says "dragging enabled", but when I go to drag the terminal point, it immediately read "dragging disabled".