fritzing/fritzing-parts #43: 78xx Voltage regulator in PCB view is incorrect

sgang007, 2016-02-06:
The generic 78xx voltage regulators in to220 tht package has an incorrect PCB view. The middle pin should be ground while the one at left will be Vin and the one at right is Vout.
Due to this bug, fritzing is automatically making wrong connections when changing to the PCB view.

[SVG image of the incorrect part](https://github.com/fritzing/fritzing-parts/blob/db2684a7410a7ca7b9c471ee657f8c7b72360d78/svg/core/pcb/sparkfun-poweric_to220-igo_pcb.svg)

[Datasheet of the mentioned part](https://www.sparkfun.com/datasheets/Components/LM7805.pdf)

aknoerig, 2016-02-14:
Thanks, but I'm not sure what you mean. When you hover over the pins you can see that they are labeled correctly. And that's the information the use should rely on, and also what Fritzing uses internally.
Or do you mean the rectangle shape of pin 1?

sgang007, 2016-02-15:
@ aknoerig : Since the pins are labelled incorrectly, the PCB layout is not being drawn correctly after I make a schematic diagram in Fritzing. 
I just made an entire power distribution circuit using several of these regulators in the schematic view and then when I moved to the PCB layout to take a printout, I just realized everything was incorrect. Then I had draw the entire circuit on hand, which was a very tedious job.

el-j, 2016-02-15:
(here is the difference between igo and adj to220 sparkfun voltage regulators)
ok i checked this out. the pinout from schematic to pcb view is consistent from my point of view. the
gnd is ground and in the middle. vin is left, vout is right. 
[image]

[image]

but we have a bug in breadboard view. there gnd is left und vin is in the middle. 

[image]

el-j, 2016-02-15:
as i see on the screenshots the adj variant is more inconsistent. this has to be fixed

sgang007, 2016-02-17:
The ground is incorrect in the PCB view also. The terminal with surrounding square should be the ground to be consistent with other parts.
You cannot see the difference because you just connected resistor which is not dependent on polarity. Try connecting two 78XX regulators with common ground and go to PCB view, you will see the connection on the left pin.

mMerlin, 2017-04-13:
Is it really the GND that is supposed to be the square pad?  I thought it was supposed to be pin 1.  Which in many, but, as shown here, not all cases is the ground.