fritzing/fritzing-parts #68: Boards with multiple header configurations

mcauser, 2016-06-29:
I have built a part which supports several header variations:
https://github.com/mcauser/Fritzing-Part-WeMos-D1-Mini

* Male headers
* Female headers
* Female headers with very long pins
* Headers above the board
* Headers below the board
* No headers

[image]

At present I have 4 separate Fritzing parts and I'm using a headers `<property>` in the .fzp file to differentiate. This lets you switch the parts by using the select field in the sidebar.

Is this the best approach?

Having 4 different parts lets me change the gender on the connectors, which makes them "connect" to breadboards. On the no-headers board, all of the pads need wires to interact with a breadboard underneath.

The 4 different parts show up in searches, which is confusing. Is it possible to make only one of the parts show up?

Ideally, I would like to maintain a single part, with the ability to toggle optional headers.
Same would apply to boards which have optional USB sockets, DC jacks, optional resistors etc.

sgparry, 2016-09-01:
I am no expert, but from my experience having 'switchable' parts (beyond what you've already done) requires code in the main app. Try asking on the forums though, you should get more responses there.

mcauser, 2016-12-07:
There has been some confusion regarding the headers (naming convention) and the PCB view.
I named the parts relative to the Breadboard view. Female above meant the far right D1 mini pictured above. The middle one is male below.

Given both sides of the WeMos D1 mini have components, the only way I can see the board being attached to a PCB is by using headers between the PCB and the D1 mini. These headers can either be female or male (pictured above), or both, in either orientation. The D1 mini is sold without any headers soldered on.

When using male headers below the D1 mini, the plastic is taller than all components, so you could mount the D1 mini to the PCB with male headers alone, soldering both sides.

The D1 mini could be mounted normal (ESP8266 on top) or inverted (CH340G on top).
Does each part require two PCB variants, normal or inverted?

sgparry, 2017-03-22:
Sorry for the long delay replying - I only just noticed this message.
try using  `<pcbView fliphorizontal="true" flipvertical="true">` in the fzp file. That should / might allow flipping in PCB view so you can mount either way up.

mcauser, 2017-03-23:
Flipping the PCB footprint, while a neat feature, doesn't really address the issue.

If you are making a "shield" that stacks on top of an Arduino Uno or a "base plate" that an Arduino Pro Mini stacks onto, the above or below footprint is the same. The fabricated PCB is going to have holes drilled where you can stack either male or female headers, above or below the PCB. The issue is not with the fabricated PCBs, it's with the app showing something meaningful on the Breadboard view. By specifying the correct gender on each hole, wires on the schematic and pcb views are connectable.