fritzing/fritzing-parts #70: part "logo" with setting copper without mask

ThomDietrich, 2016-06-30:
When creating a logo text element on the pcb using the copper option, etching is calculated accordingly. Solder mask is however not left out, so the copper will be covered and potentially hidden. Is / Shouldn't there be an option to unmask the copper logo? (this is of course a visual more then a technical matter)

ThomDietrich, 2016-08-18:
[image]

ThomDietrich, 2016-08-18:
In my understanding, `silkscreen` needs to be added as a svg group element. I did however not find the part "/resources/parts/core/breadboardlogotext.fzp" Can somebody please help fix this small bug ?

ThomDietrich, 2016-08-24:
I would take on this task if someone gave me a starting point, some info to work with... @el-j ?

el-j, 2016-08-25:
hej thom, 
at this point i am not sure how to solve the problem. the resources files are in the source code. there you can find the breadboardlogotext.fzp. just look into the git repo in: https://github.com/fritzing/fritzing-app/tree/master/resources/parts/core

please notice, if you want to change anything in the resource files, you have to compile fritzing from source to see the changes. maybe you could make an extra "logo"-part for testing purpose. 

your idea, adding a silkscreen layer simply adds a print of the logo but do not let out the white color...i think therefore the logo has to be some kind of connector but than it will be silver and not copper... like a smd pad. 

in conclusion: geting a copper-color for the logo is nearly impossible, a silver one should be possible if the logo itself gets a connector.

if you want you could try to find a way adding the logo as a "connector"

JoshuaLowe1002, 2018-04-02:
Any updates on this?? I am trying to add copper text to show on my PCBs too

ThomDietrich, 2018-04-02:
I would still be interested as well. Sadly I can't offer any help with the implementation...

ai91, 2022-07-10:
4 years later... No updates so far?
Does nobody interested in the feature? Or is there any common workaround everybody use?

failiz, 2022-08-02:
I think that as @el-j said, the part needs also to have a connector like the pad part. I would create a new part. In any case, I strongly recommend you to ask this in the forum as they can probably help you there.

ThomDietrich, 2022-08-02:
To be clear, the ticket is not about copper or silver, or parts. 
It's about removing the silkscreen where text is planned in the PCB design. Imho this is an option Fritzing should offer via its UI -> This ticket is a feature request :)