fritzing/fritzing-parts #77: large qty of contribs

ladyada, 2016-08-28:
hiya, this was a "summer of fritzing" for us - we did a sweep and created about 330pixel-perfect custom generated parts from all the Adafruit designs. you can see some of our output https://blog.adafruit.com/?s=fritzing

Right now they live in our repository https://github.com/adafruit/Fritzing-Library but they'd also make a great addition to the current Fritzing parts collection (there are a few Adafruit parts in there right now but we've redone those as well)

Our question is: what is the best way to submit this many parts? should we submit the 6 sub-bins? do y'all want to make the bins and we can do a huge pull request? thanks, its been quite a journey, we'll be keeping up to date going forward with approx one new fritz a week :)

sgparry, 2016-09-01:
I am really excited to see these, you guys make some really awesome parts. No expert but you could just include the new bin definitions in the pull request - I added a new bin for Dagu parts...

aknoerig, 2016-09-12:
Wow, what an awesome and huge contribution. Well done! Especially using the eagle2fritzing script will make sure they are 100% workable.

Given that there are so many parts, I think it's justifiable to have them spread over 6 bins. We'll need to find a better UI for this many bins eventually, but I think most people are using the search anyway.

So either make it a pull-request or I can also make you a contributor to the repo. What do you prefer?

ladyada, 2016-09-12:
Hiya, i can do a pull request. 
i'm not sure how to set up bins for Fritzing that will show up properly (since we manully import) - right now we have an fzbz for each bin...?

aknoerig, 2016-09-12:
Perfect! The bins are just an .fzb and a .png icon and go into `fritzing-parts/bins/more`. That's it. See the other files there for an example.

aknoerig, 2016-09-13:
Maybe I should add that the .fzbz is just a zip file and contains the .fzb. So you can just rename the .fzbz to .zip, extract it, and find your .fzb inside.

el-j, 2016-09-13:
hej, 
thanks for all the parts. i think the problem is that we cannot use fzpz at the moment. 
so we need the open files for the parts in the fritzing-folder structure. there is a python script that can clean fzpz and separate the files to the folder-structure fritzing needs. so at the moment a pull request with fzpz and fzbz cannot be integrated on the fly, to the fritzing-parts repository. 

the other thing is the checking of the parts. we have to make sure that they are working in all 3 views and that the gerber export is correct. 

thanks for the great work so far!

KjellMorgenstern, 2020-10-28:
This looks more like a feature request in Fritzing: Add the ability to include 3rd party part repos.

mMerlin, 2020-10-28:
Linking to 3rd part repos would be one solution. The way I read the original post though, this was a request for information on how to directly add (and update) the parts from the adafruit repo into the Fritzing parts library. Which would be a PR that added all of the fzp and svg files, plus the changes to create and update bins.

KjellMorgenstern, 2025-09-17:
We are integrating Adafruit parts for upcoming Fritzing 1.0.6.

**using contrib folder**
Parts will be put into the "contrib" subfolders instead of the "core" subfolders.
In Fritzing 1.0.6, contrib parts will always be bundled when saving a sketch (core parts are not, because Fritzing always includes them)
This is an important change: Because the parts are bundled with sketches, future release of Fritzing can simply replace these parts if we have any fixes for a part. It is also the same way Fritzing handles the part if it would have been imported by the users direclty as fzpz.
This should make it much easier to maintain parts. Eg. We can just re-import them from the external repo to account for upstream changes, without having to worry about compatibility with previous Fritzing releases.

In extrem cases, when we really want to notify about broken parts, the obsoletion mechanism using the "replaced_by" attribute should still work.

**all parts in one bin**
I think a flat structure with one bin will work best.
Full text search, tags ( https://github.com/fritzing/fritzing-app/issues/4258 ) and family should account for this.
[image]

To categorize the parts within the bin, I loosely used the categories listed here: https://www.adafruit.com/categories
The called an LLM to categorize parts by their descriptions and names. The result is stored in a text file, that maps module ids to categories. We can edit this file manually to adapt categories, here a first draft with ~80 parts (the ones most recently added to Fritzing-Library) : 

[categories_map.txt](https://github.com/user-attachments/files/22379094/categories_map.txt)

**performance**
Loading a bin with hundreds of parts currently is very slow. We have a plan for that (storing the icons as blobs in Fritzings internal db, instead of reading xml and svg for each icon. Partly done, but not for 1.0.6)

**part fixups**
We apply some fixes on the fly, like replacing "OCRA, OCRA" with "OCR-Fritzing-mono" . The former would result in a random system font replacement anyway, so replacing it with Fritzings own fonts seem save. Similar for Noto Sans (DroidSans, OpenSans)
For these fixes, we currently add a _xy suffix to the module ID, to not interfere with the orignal parts from Adafruit Fritzing-Library. Many parts already contain a partnumber, but didn't use the part number property for this. This is also adapted, so that part numbers show up in the BOM.

**the bin icon**
Bin icons colorize when selected. I tried a soft pink color, like often seen on the Adafruit webpage, but that lead to bad contrast. Therefore I used a color inspired from the QtPy ESP32 
[image]
Please let me know if you prefer different bin icons.