fritzing/fritzing-parts #81: gh-pages merge conflict between master

paulvollmer, 2016-11-18:
i've try to merge the latest commits from the `master` branch into the `gh-pages`, but there is a confrict. the `gh-pages` are a lot of commits behind the master and...

here the comparison: https://github.com/fritzing/fritzing-parts/compare/master...gh-pages

to fix the issue we can rename the current `gh-pages`branch and create a new branch from the master.

@el-j there are come commits at the gh-pages you've made and i think it is only some typo fixes at the main README.md. can you manual merge it please...

cheers, paul

el-j, 2016-11-18:
i assume the gh-pages branch has the complete old history of the parts. 
so deleting the old one and create a new seems ok. the gh-pages branch is not used from others at the moment

el-j, 2016-11-18:
i have made a master merge just at the moment. gh-pages should be clean now

closed 2016-11-18 by el-j

paulvollmer, 2016-11-19:
ok, great