fritzing/fritzing-parts #85: The latest parts update has broken the pcb footprint for the Camdenboss_CTB0158-2_5_08mm_pitch_90deg_terminals_pcb.svg

vanepp, 2017-02-11:
The latest parts update has broken the previously working footprint for the Camdenboss_CTB0158-2_5_08mm_pitch_90deg_terminals_pcb.svg
file. There is a topic in the forums looking for a fix, and it was working for me until I checked updates and loaded the latest updates at which time I too became broken.

el-j, 2017-02-11:
i am really sorry, i have to fix all the pcb-svgs because of many transform/translate properties, that causes trouble if the part is rotated and getting exported.

sadly i override the  Camdenboss_CTB0158-2_5_08mm_pitch_90deg_terminals_pcb.svg with the CTB0158-6 footprint.

fixed with new update

closed 2017-02-11 by el-j

vanepp, 2017-02-11:
Wow, fabulous service, thank you! You have my sympathy on the translates, Inkscape loves them and I find them evil and hard to eliminate (but I'm still learning to make parts).

commit cea68dce (2017-02-11): fixes #85 removed false connectors from camdenboss ctb0158-2