fritzing/fritzing-parts #107: Raspberry Pi Zero PCB part completely wrong

gbaman, 2017-09-28:
I have discovered the hard way this evening after getting a set of boards sent off and manufactured, that the Raspberry Pi Zero PCB part in Fritzing is completely wrong. The headers are completely the wrong size... Rendering the boards completely useless. It seems like 2mm spacing instead of normal 2.54mm spacing between the pins.
Note the the normal Raspberry Pi model has the correct spacing, just not the Pi Zero part.

Although I hand checked a number of the parts on the board before sending it off, checking a core Fritzing part was not something I expected to have to do?

[image]

linuxlurak, 2017-09-29:
I switched to diptrace. It's the way to go... No offense!

paulhayes, 2018-01-03:
This almost caught me out. Luckily I spotted it. Replaced my part with the beautiful one posted here. I hope this user contributed replacement gets incorporated, it's got a better schematic layout too.

aknoerig, 2018-01-03:
Fixed via pull request 114

closed 2018-01-03 by aknoerig