fritzing/fritzing-parts #126: DC power plug has smaller PCB VCC hole than normal

rfinnie, 2018-02-02:
The Power Plug (DC jack) part has three holes, two GND and one VCC.  All three are approximately 2.6mm diameter in PCB view.  However, the "standard" DC jack's spades are 2.5mm for the GNDs and 3.0mm for VCC.  See http://www.cui.com/product/resource/pj-002a.pdf for an example, but I haven't been able to find a jack supplier with 2.5mm for all three.  The VCC hole should probably be 3.1mm.

KjellMorgenstern, 2020-10-28:
Indeed it looks like it is 1x3 and 1x3.5 , according to the datasheet of the pj-002a and many similar power jacks.

[image]

KjellMorgenstern, 2020-10-28:
As a workaround, it might be possible to use the sparkfun power jack (also included in Fritzing), which has 3mm drill holes

KjellMorgenstern, 2020-11-08:
I think the "standard" plug used in Fritzing matches this component:

https://asset.conrad.com/media10/add/160267/c1/-/en/001898861DS01/datenblatt-1898861-wuerth-elektronik-niedervolt-steckverbinder-buchse-einbau-horizontal-550-mm-205-mm-1-st.pdf