fritzing/fritzing-parts #160: Parts update appears to be broken

vanepp, 2018-08-08:
It appears that parts update is broken, presumably because of a change in the repository. On 3 different machines Fritzing help->Check for updates
gets the message

Unable to access network site for 'c:/Fritzing/fritzing.0.9.3.b.32.pc/fritzing-parts'.

Fritzing can not update the parts folder automatically. 

Peter

el-j, 2018-08-08:
hello peter,
that seems odd. i tested it with a fresh downloaded version of fritzing and made the parts-update check. 
the download and the update works for me with 64bit version of fritzing 0.9.3b.

do you maybe did some git-business in your parts-folders on your own? or changed parts inside of it?
this could be a reason for a problem with a new parts-update.

vanepp, 2018-08-08:
Interesting! Must be something local screwing me up. While two of the three machines are used for development (and thus may have a corrupted parts repro), the third is a little used laptop that doesn't have any changes and it does the same thing and used to work (at least some number of months ago). I'll try a new install and see if I can figure out whats wrong here. I'll close this ticket as it looks to be a false alarm.

Peter

closed 2018-08-08 by vanepp