fritzing/fritzing-parts #163: RioRand LM2596 has wrong pin location

ranjib, 2018-08-27:
This is already reported, http://forum.fritzing.org/t/correction-for-the-riorand-lm2596-from-core-somewhere/2272
But I am still encountering this error, I am using 0.9.3b.

I was also not able to join the forum ans respond to the above thread due to captcha error (forum signup was not possible due to captcha service being down)

vanepp, 2018-09-25:
Hmm, the fix in the forum article was committed to core. Assuming you are running on Windows, the problem likely is that parts update isn't working. The fix (until the issue with libgit2 gets resolved) is to manually clone the parts repository in to the directory fritzing-parts in the fritzing install directory and then rebuild the parts database to get the latest fixes. I've sent a message to who I think is the web site maintainer about the forum captcha service being down and we will see if anything happens.

Peter

KjellMorgenstern, 2020-10-28:
Already fixed

closed 2020-10-28 by KjellMorgenstern