fritzing/fritzing-parts #164: Parts update on Windows is broken

vanepp, 2018-08-31:
It appears that a change in the repository has broken parts update on Windows but not on Linux. I have finally managed to get a development environment apparently working on Windows in that it will correctly load the part repo from the 9.3b release, then update it from the github repro and refuse to load the new repro on the fritzing restart:

Original Fritzing parts repository (from the 9.3b zip file):

QT Creator console output:

20:56:50: Starting F:\fritzing_dev\debug64\Fritzing...
"translation en_CA loaded 0 from ."
"registering font family: Droid Sans DroidSans"
"registering font family: OCRA OCRA"
"got transaction"
"referenceModel::loadAll completed"
"parts count 1979"
"load something"
"checking for backups"
"check files to load 0"
"create empty sketch"
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
"loading bin 'Core Parts'"
"module id mystery_part_sip_3_100mil not found in database"
"module id generic_ic_dip_8_300mil not found in database"
"module id generic_female_pin_header_2_100mil not found in database"
"module id generic_shrouded_pin_header_10_100mil not found in database"
"module Mystery Part - 3 pins:mystery_part_sip_3_100mil obsolete svg F:/fritzing_dev/fritzing-parts/svg/obsolete/icon/mystery_part_icon.svg"
"module Generic female header - 2 pins:generic_female_pin_header_2_100mil obsolete svg F:/fritzing_dev/fritzing-parts/svg/obsolete/icon/generic_female_header.svg"
"Setting platform to Arduino"
"Setting board to Arduino UNO"
"Setting board to Arduino UNO"
"Setting port to COM1"
"Setting platform to Arduino"
"adding layer kin 57471 3 25 130f4660"
"adding layer kin 57472 3 18 130f4d50"
"add part ti:'Rectangular PCB - Resizable' id:57470 it:'PCB1' vid:3 vlid:16 spec:0 x:0 y:0 z:1.5 flg:0 gi:421f990"
"add part ti:'Rectangular PCB - Resizable' id:57470 it:'PCB1' vid:1 vlid:-842150451 spec:0 x:0 y:0 z:0 flg:0 gi:421d0f0"
"add part ti:'Rectangular PCB - Resizable' id:57470 it:'PCB1' vid:2 vlid:-842150451 spec:0 x:0 y:0 z:0 flg:0 gi:42205c0"
"add part ti:'RSR 03MB102 Breadboard' id:57490 it:'Breadboard1' vid:1 vlid:1 spec:0 x:0 y:0 z:1.5 flg:0 gi:146fdd70"
"add part ti:'RSR 03MB102 Breadboard' id:57490 it:'Breadboard1' vid:2 vlid:1 spec:0 x:0 y:0 z:-1 flg:0 gi:146fe040"
"add part ti:'RSR 03MB102 Breadboard' id:57490 it:'Breadboard1' vid:3 vlid:1 spec:0 x:0 y:0 z:-1 flg:0 gi:146fe310"
"finish up sketch loading"
"http://fritzing.org/download/feed/atom/windows-64bit/?version=0.9.3b.04.19.5c895d327c44a3114e5fcc9d8260daf0cbb52806&sysname=Windows&kernname=NT&kernversion=6.1&arch=x86_64&sysversion=7%20&pid=51db8e01fc292e4f5e9c8c63ea42e14cF%20Server%202008%20R2%8"
"http check new version"
"http check new version no error"

git log

commit 667a5360e53e8951e5ca6c952ae928f7077a9d5e (HEAD -> master, tag: 0.9.3b, origin/master, origin/HEAD)
Merge: 983b07b 6e4c8c3
Author: el-j <bluearc.in.sky@gmail.com>
Date:   Thu Jun 9 13:53:56 2016 +0200

    Merge branch 'master' of https://github.com/fritzing/fritzing-parts

commit 983b07b58a39c473c7aea907d98840c482be8768
Author: el-j <bluearc.in.sky@gmail.com>
Date:   Thu Jun 9 13:53:48 2016 +0200

    fix:#3225 arduino uno

Current fritzing-parts repro (via update to the above from Fritzing) which breaks:

git log

commit f5e9779b1996cb6b5132b5bccdd281e1994f3bbe (HEAD -> master, origin/master, origin/HEAD)
Author: H. Phil Duby <philduby@phriendly.net>
Date:   Fri Jun 22 21:50:39 2018 -0600

    cleanup obsolete family text prefixes

commit 5440a4d5d17b3faa70a19079e8241e390d660601
Author: H. Phil Duby <philduby@phriendly.net>
Date:   Fri Jun 22 19:18:26 2018 -0600

    cleanup family and other tags for microsdmodule.fzp

QT Creator console log:

21:06:11: Starting F:\fritzing_dev\debug64\Fritzing...
"translation en_CA loaded 0 from ."
"registering font family: Droid Sans DroidSans"
"registering font family: OCRA OCRA"
"got transaction"
"referenceModel::loadAll completed"
"parts count 2134"
"missing connector in STM32F103C8T6_blue_mini_db4c63b0e434949d3cc651e6f45c9298_3"
"load something"
"checking for backups"
"check files to load 0"
"create empty sketch"
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
"loading bin 'Core Parts'"
"module id mystery_part_sip_3_100mil not found in database"
"module id generic_ic_dip_8_300mil not found in database"
"module id generic_female_pin_header_2_100mil not found in database"
"module id generic_shrouded_pin_header_10_100mil not found in database"
"module Mystery Part - 3 pins:mystery_part_sip_3_100mil obsolete svg F:/fritzing_dev/fritzing-parts/svg/obsolete/icon/mystery_part_icon.svg"
ASSERT: "!"Tspans should be drawn through QSvgText::draw()."" in file c:\users\qt\work\qt\qtsvg\src\svg\qsvggraphics_p.h, line 224
QWaitCondition: Destroyed while threads are still waiting

It looks like the changes to obsolete parts in the new repo may be at fault but I don't know how to back off the commits using git in order to figure out what part is causing the problem.  I tried (in the new parts repo)

git reset 2cd1737be3bef3b7ef65b5bca4c8f9964035be43 commit

which I think should revert to the same state as the 9.3b repo but a diff of the repository indicates it has not worked (and indeed Fritzing still breaks) as the two repos are not the same. If someone can explain how to back off the commits one at a time I will try and figure out what part is causing the problem (and probably why there is a problem). Meanwhile I will continue poking at git documentation to try and figure out how you are supposed to back off changes. 

Peter

vanepp, 2018-09-01:
Unfortunately I think the above is a red herring in the form of an issue with the new version of QT I'm using. I think the actual problem on 9.3b is likely that the libgit2 it uses no longer has any cyphers that github will accept. The actual error message from 9.3b is "Unable to access network site for 'C:/fritzing.0.9.3b.64.pc/frtizing-parts' Fritzing cannot update the parts folder automatically." At least 2 other people on Windows in the forums have the same issue as well. If I load the copy of the parts repo that my development version won't load in to 9.3b it appears to load just fine indicating the problem isn't in the repo but likely in libgt2. I don't have a debug version of 9.3b to test on (and can't make one because I would need an old version of the libgit2 library which I don't know how to make). If someone has a still working development environment for 9.3b perhaps they could try it and see if they can identify the problem. A workaround would appear to be to manually download the parts repo in to the local file. 

Peter

el-j, 2018-09-02:
thanks for the report
1. just to clarify: the problem is not in the prats-repository?
2. the problem appears with a "self-build" fritzing version?
3. the issue could be libgit2? did you try to build it from the instructions?
https://github.com/libgit2/libgit2

otherwise i could share a win32+64 build of libgit2 what i builded a while ago, but it is 
for windows. can you use this?

vanepp, 2018-09-02:
1) Correct, the problem with 9.3b doesn't appear to be the parts repository. A build of head with the latest QT appears to have an issue with some parts causing an assert to trip that I'll poke at later. 

2) The problem is with the 9.3b release version. At some point in the past 6 months (don't know exactly when) parts update has stopped working. That appears to be that it (on Windows only, Linux and probably the Mac still work fine, although I don't have a Mac to test on) can't make an ssh connection to github.con. I suspect that the cyphers in the libgit2 distributed with 9.3b have gotten too old and github rejects the connection (because tcpdump indicates the connection dies during handshake). 

3) A current version of libgit2 built on Win7 with the latest patches (and thus probably a later set of cyhers) works fine and will update the parts repo correctly. The distributed 9.3b zip file on Windows apparently can't make the connection to github and will not update the parts repository. I suspect the only fix available is to recompile 9.3b (or head) with a new libgit2 build. If someone has the source library (the include files and dlls, the distribution zip only has the dlls not the includes) then I could maybe verify that the failure is because of old cyphers, but the solution is still likely to be a new release of Frtizing which may or may not be possible to arrange. A workaround (which may be all we have) is to manually download or clone the parts repo and install it in the 9.3b  parts directory. Doing that appears to update the parts just fine. 

Peter

vanepp, 2018-09-13:
The problem appears to be the version of libgit2 in use. The current version appears to use tls1.1 (which I expect github is no longer supporting). libgit2 v0.26.3 (which has this in the change log: "Enable Windows 7 and earlier to use TLS 1.2") works from a compiled from source version of fritzing, but just substituting the dll in to the original zip file still doesn't work (possibly because internal flags need to change). In any case not a parts repo issue so I will close this and open on in the fritizing-app repo instead. 

Peter

closed 2018-09-13 by vanepp