fritzing/fritzing-parts #166: DB25M Pin 14 Does not match Schematic

pgodwin, 2018-12-17:
It looks like Pin 15 is swapped with Pin 14 for the DB25M part between Schematic and Breadboard view.

Using v 0.9.3 - updated all parts

Example Schematic - Pin 14 to 5V, Pin 15 to GND via 220ohm resistor:
[image]

Switch to breadboard:
Resistor is connected to Pin 14 rather than 15:
[image]

Example:
[fritzing wrong pin.zip](https://github.com/fritzing/fritzing-app/files/2682526/fritzing.wrong.pin.zip)

Originally raised in https://github.com/fritzing/fritzing-app/issues/3431

pgodwin, 2018-12-17:
Confirmed in the parts editor. Pin 14 on schematic is definitely mapped to pin 15:
[image]

KjellMorgenstern, 2020-02-21:
The issue itself is easy to fix, but the part obsoletion process must be verified for this case where only the svg is changed.

pgodwin, 2020-02-25:
Thanks @KjellMorgenstern  - anything you need from me?