fritzing/fritzing-parts #171: Calliope Mini

el-j, 2019-02-08:
@HaraldRau can you please share your work with the Calliope Mini you have done and i can assist on that?
please write an email for all the details. (mail in my profil-description)

HaraldRau, 2019-02-09:
I tried to build a part for Fritzing. It's the [Calliope Mini](https://github.com/calliope-mini/calliope-mini.github.io). I could not realize all connections yet. Can somebody help me with it?
Can I upload my version somewhere?

Thank you!
Harald!

el-j, 2019-02-10:
ok now i did a first version of the calliope mini rev 1.3 based on the 
[original brd file](https://calliope-mini.github.io/assets/v10/eagle/Calliope%20mini%20rev1.3.brd)

checkout the nightly branch and rebuild the database to use:
1. go to fritzing-parts-folder
2. checkout nightlyParts branch
` git checkout nightlyParts`
3. start fritzing and regenerate the parts db.

`https://github.com/fritzing/fritzing-parts/tree/nightlyParts`

el-j, 2019-02-10:
[image]
[image]
[image]

el-j, 2019-02-10:
[CalliopeMiniRev1.3.fzpz.zip](https://github.com/fritzing/fritzing-parts/files/2848949/CalliopeMiniRev1.3.fzpz.zip)

here is the exported part. please remove the zip file-ending after download.
(was needed to do the upload here :) )

HaraldRau, 2019-02-10:
Thank you very much,
it works perfect.

`git checkout nightlyParts` must run in a Termimal?

I will install git an run the command.

HaraldRau, 2019-02-10:
it doesnt work. I have no Git-Repository. My Version in an portable Version LINUX 64 bit.

No matter, the imported part work perfekt, manny thanks!

Harald!

el-j, 2019-02-10:
yes, u can go to the fritzing-parts folder of the fritzing app. this should be a git repository. so you can pull from https://github.com/fritzing/fritzing-parts

CAREFUL: after changing the branch you will not be able to use the parts update from the application, and you have to rebuild the parts database.

closed 2019-02-10 by el-j

HaraldRau, 2019-02-10:
...to early, the part crash when i build a autorout for pcb.

Who can i upload my experimental part?

el-j, 2019-02-10:
i cannot confirm on windows 0.9.3.b.
please attach the sketch here
(put .zip at the end of the filename to upload here)

el-j, 2019-02-10:
> it doesnt work. I have no Git-Repository. My Version in an portable Version LINUX 64 bit.
> 
> No matter, the imported part work perfekt, manny thanks!
> 
> Harald!

u can delete the fritzing parts folder and do a 
`git clone https://github.com/fritzing/fritzing-parts.git`

HaraldRau, 2019-02-10:
[Calliope Mini mit PIN Header.fzpz.zip](https://github.com/fritzing/fritzing-parts/files/2849020/Calliope.Mini.mit.PIN.Header.fzpz.zip)
Here is my Part.

el-j, 2019-02-10:
there is no problem on my side
[image]

el-j, 2019-02-10:
please open a new issue with sketch and precise failure description.
thanks

HaraldRau, 2019-02-10:
Here is the screen:
**Breadbord:**

[image]
**PCB:**

[image]

When i klick "Autoroute" crash the App without errors message.
I will run in a Terminal and check the sterrer.

HaraldRau, 2019-02-10:
The Error Message is:
`free(): invalid pointer
Aborted (core dumped)`

HaraldRau, 2019-02-10:
I tried with a other part, the same result.

el-j, 2019-02-11:
ok, i can confirm now. -... strange. i check it

el-j, 2019-02-11:
ok it was because of the pads. there were on bottom and top so it broke ...
please try to checkout the nightlyParts and try again.

HaraldRau, 2019-02-12:
O.K., I want to test it. I will build a development environment. At moment i'm testing in my production environment. That's not a good idea.

HaraldRau, 2019-02-13:
i do:
`git checkout nightlyParts`
and get:
`Branch "nightlyParts" folgt nun Remote-Branch "nightlyParts" von "origin`
and regenerate the Part Dbase!
I cant find the Calliope?
How can i find the Part? I search for "Calliope", but no result!

el-j, 2019-02-15:
ok so: 
`git checkout <name>` 
will change the git branch you are on
u can now use
`git fetch` <--- will read all changes and branchnames from git but will not download anything
`git pull` <--- will download all the changes from the git repository  
so, you have to "pull" to get the last changes.
after that do the regeneration of the parts lib again
and you should find the calliope by its name ;)
this is a nice [cheat sheet](https://services.github.com/on-demand/downloads/github-git-cheat-sheet.pdf) for git 🗡