fritzing/fritzing-parts #194: Relais maluska dil in V5 (only) mode can not be selected

ChaosLord76, 2019-06-29:
i just discovered, that when i select the maluska relais and switch to 5V mode, you can not select the device. where ever you click on it fritzing starts creating a wire connected to the common rail. When you switch to any other option (i.E. 5V - max 96 VDC) everything works fine.

Steve

KjellMorgenstern, 2019-06-29:
Just reproduced the issue

[image]

Indeed, the part can not be moved on the breadboard.

vanepp, 2019-09-30:
The problem here isn't the maluska relais part but the family. The family is relay which includes the Omron1 power relay which has a different pin out. Because the family is the same, Fritzing believes they are identical and tries to substitute one for the other, with the result you see (too few contacts in the Omron1 part causing the red background to indicate connector problems.) The solution would be to change the family name of the maluska relais part so it won't try and substitute a relay that doesn't have sufficient pins. In the longer term we need documentation and preferably automatic checking for the family to make sure all members have the correct number of pins.

closed 2023-06-30 by KjellMorgenstern