fritzing/fritzing-parts #202: DS18B20 pinout wrong?

PeterJones7, 2019-07-15:
The DS18B20 part seems to have the wrong pinout - GND and VCC are reversed in breadboard view I think.

vanepp, 2019-09-30:
It looks correct to me. The data sheet is somewhat confusing in that it is showing the to92 case in bottom view (which is reverse to how Fritzing shows it). Breadboard matches the top drawing in the data sheet for the to92 (which is top view that Fritzing uses), ground (pin1) on the left. Connecting a wire there in breadboard connects to ground on both schematic and pcb as it should.

KjellMorgenstern, 2019-10-01:
There are many illustrations of this part in bottom view, I think that might have caused the confusion. Also, in Breadboard and PCB view, the part is rotated by 180 degree by default, but if you watch at the silkscreen, and that it is in top view, you'll see it is correct.

miriki, 2020-10-08:
I'd like to affirm PeterJones7 - If you place a DS18B20 into the breadboard view the flat side is directed to the front / user (toward bottom of the monitor, while the rounded back side is directed to the back, the top of the monitor). The left pin is called VDD, the right one is GND. If you connect it that way the sensor will get hot.
[image]
The left pin is GND and the right one is VDD, at least the ones in my drawer follow that.
[image]

KjellMorgenstern, 2020-10-08:
What is the connector name when you mouse hover on the part in breadboard view?
Trying this (on my dev version of Fritzing) left is GND, and right is VDD same as the physical one from your drawer.

[image]

mMerlin, 2020-10-08:
I get the same results as [Kjell](https://github.com/KjellMorgenstern) for breadboard view pin order (GND on the left) using Fritzing version 0.9.4 with an up to date parts library. I do not get the id in Inspector though. That must be new in the dev version.
[image]
Depending how Fritzing was installed, it is quite possible that the part library can not be updated automatically. The app might not have write access to the folder the parts library is stored in. You can try selecting "Check for updates" on the Help menu (for recent Fritzing versions), and see what that tells you. Most failures should show the path to the parts library folder.

miriki, 2020-10-08:
I am using version 0.9.3 beta of the Fritzing software with a download made on 31-Aug-2020
[image]
There is no update opportunity - the "close" button stays grayed out and the requester has to be closed via the [X] button.
[image]
The whole computer is newly set up (Win 10 from scratch) roughly 4 weeks ago.
Without connected wires the sensor shows at hovering
[image]
And the inspector shows unf. no specific id
[image]

mMerlin, 2020-10-08:
A download done at the end of August 2020 should have gotten the 0.9.4 version. That definitely looks like you are working with an old version of the parts library (as well as application version).

The symptoms don't seem to be quite right, but have a look at [this FAQ](https://github.com/fritzing/fritzing-parts/wiki/FAQ:Fritzing-Freezes-on-Start). Since this is a recent/fresh install of Fritzing, it could that the initial update of the parts library got aborted, corrupting things is such a way that updates are not happening. Or, since this is the first update (which is big), you are killing it before it finishes.

"update information could not be obtained" or "no update information could be retrieved" does not say that there is NOT an update. It seems to say that it could not be found or downloaded. Which is another indication that the parts library you are using is out of date.

miriki, 2020-10-09:
Right when we're talking about... I remember there was an issue with the "please pay and download" button and thus I downloaded from another source, getting the 0.9.3 instead of the 0.9.4. I just downloaded the 0.9.4 a few minutes ago and replaced my old version with it.

Bingo! The pinout of the DS18B20 shows GND when hovering over the left pin and VDD at the right pin.

The parts file DS18B20.fzp under ./fritzing-parts/core/ shows 10-Jun-2016 in the old and 01-Dec-2019 in the new version.

So, the error is up until the 0.9.3 version (including) and why that thing isn't able to update the library, I don't know.

Problem solved, many thanks!

KjellMorgenstern, 2020-10-09:
Fritzing 0.9.3b is not yet using transport layer security. Probably github (from where the parts are pulled) stopped supporting this. That means new installs of 0.9.3b won't get any updates, and will be stuck with the 2016 version of the parts. Raising the priority for 32 bit support in 0.9.5 even more, but that is on a different issue.

mMerlin, 2020-10-21:
A possible work around

Would a manual git pull work for this case? That is, is the 2016 parts library installed with 0.9.3b a true git repository, with origin pointing the github repository?

KjellMorgenstern, 2020-10-23:
@mMerlin Yes, it should.

vanepp, 2020-10-30:
After doing the pull, you then need to regenerate the parts library to update the data base via (from memory, I don't have Fritzing here) Parts->regenerate parts database which will do the regen and then shutdown Fritzing (the same as a parts update does.) 

Peter