fritzing/fritzing-parts #203: Adafruit HUZZAH ESP8266 breakout board Gerber file fail

chrisgimson, 2019-07-18:
Exporting to a gerber file produces a solid copper block on the top layer underneath the component that shorts out all of the pins!.  You do not see this on the Fritzing PCB view nor on exported image files - only on the gerber file which you will be blind to unless you check the gerber files with a viewer.
The other ESP8266 parts in the library do not have this problem so the short term fix is to simply avoid the adafruit part if you are using gerber files.

KjellMorgenstern, 2020-10-27:
The part has a silkscreen component inside the copper1 component
```
<g id="copper1">
...

                    <g gorn="0.1.0.0.0.0.6"  id="silkscreen">
                            <line x1="216.135" x2="119.471"  fill="none" stroke-width="1.1114" stroke="#FFFFFF" y1="446.456" y2="446.456"/>
                            <line x1="119.471" x2="119.471"  fill="none" stroke-width="1.1114" stroke="#FFFFFF" y1="446.456" y2="500.611"/>
                            <line x1="119.471" x2="216.135"  fill="none" stroke-width="1.1114" stroke="#FFFFFF" y1="500.611" y2="500.611"/>
                            <line x1="216.135" x2="216.136"  fill="none" stroke-width="1.1114" stroke="#FFFFFF" y1="446.456" y2="459.806"/>
                            <line x1="216.135" x2="216.135"  fill="none" stroke-width="1.1114" stroke="#FFFFFF" y1="487.25" y2="500.611"/>
                    </g>
</g>
```

KjellMorgenstern, 2021-09-16:
Solved by 17fbf390930

closed 2021-09-16 by KjellMorgenstern