fritzing/fritzing-parts #206: Arduino Yun, SPI Pins wrongly mapped to D13,D12,D11

khawajamechatronics, 2019-07-18:
Hi,

I was using Yun in a project, As Yun uses Atmega32u4 whose SPI pins are on ICSP header not on D13,D12,D11.

But When I wire the ICSP to SPI Module, I see D13,D12,D11 highlighted, which should not be.

Please correct.
Thanks

KjellMorgenstern, 2020-10-27:
While this is correct, I tend to close this 'won't fix' because the Yun has been discontinued.

HaraldRau, 2020-11-04:
@KjellMorgenstern 
Hello, the connections are wrong in the schematic and breadboard.
Should I correct it?
[image]
[image]

HaraldRau, 2020-11-04:
I removed the wrong connections to the ICSP header. But there is no ICSP header on the board. I will insert the ICSP.

Here the first result:
[Yun_Arduino_V_1.fzpz.zip](https://github.com/fritzing/fritzing-parts/files/5489631/Yun_Arduino_V_1.fzpz.zip)

KjellMorgenstern, 2020-11-04:
Thanks, I will create a PR from the zip file

HaraldRau, 2020-11-05:
I added the ICSP header. I used the pcb from the Arduino Uno. Here is the zip-file!  I checked the files with "FritzingCheckPart-master", but cannot evaluate the output.
[Arduino_Yun_ICSP.fzpz.zip](https://github.com/fritzing/fritzing-parts/files/5492524/Arduino_Yun_ICSP.fzpz.zip)