fritzing/fritzing-parts #207: core part "Electrolytic Capacitor" has + - pads swapped.

wile3, 2019-07-18:
The + side should be the square pad, and the - side should be the round pad for PCB view.  It is currently reversed in 0.9.3.

benreu, 2019-09-03:
Could you provide a link or specification for this? Also, is changing the capacitor going to affect (and break) PCB's already designed?

vanepp, 2019-09-30:
"Also, is changing the capacitor going to affect (and break) PCB's already designed?"

No, should we choose to do this (and I currently don't see a reason to do so) the obsoleting mechanism (once we figure out how it works) would protect the older sketches. It keeps the old part in the obsolete bin and offers to update the sketch. I believe it can reroute the pins correctly if told to do so, but there is no documentation other than the source that I am aware of. Once I finish updating  the FritzingCheckPart script I intend on trying to document part obsoleting.

wile3, 2019-10-01:
I can't find the reference from when I reported this issue 2 years ago. I found that from multiple sources at that time before reporting it. From my current research, it sounds like this is more of a preference thing, than a standard. You can close this issue as it is not necessary to change.

KjellMorgenstern, 2019-10-01:
We did not find a standard of how this is done. In Fritzing, the square on the PCB marks the negative side.
However, on LEDs the square marks the anode in Fritzing, which is not common, see for example
https://electronics.stackexchange.com/questions/210753/whats-the-standard-for-denoting-the-orientation-of-an-led-on-a-pcb
Normal diodes have the square on the negative side in Fritzing.

So, now I think the PCB view for LEDs should be changed.

closed 2020-10-27 by KjellMorgenstern