fritzing/fritzing-parts #216: Something has gone wrong with the latest parts commit.

vanepp, 2019-08-30:
The latest part commit for the frc parts appears to be broken somehow. On 0.9.4 prerelease on Win7, it discovered a parts update and listed a bunch of placeholder files to delete (normal I expect) but the change log shows no frc parts but a bunch of the previous commits I think (calliope and Hive5 which are already in as far as I know). Completing the parts update appears to work but the frc parts and the new bin are not there. Manually cloning a copy of the parts repo and replacing it in the 0.9.4 directory, then clearing the user directories, starting fritzing and doing parts->Regenerate parts database causes the usual shutdown and on restart, the frc bin and the new frc parts are present. So it appears there is a problem in the update mechanism somewhere. Fritzing 0.9.3b on Win7 has the bad libgit2 version and thus won't automatically update, but 0.9.3.b on Linux (Ubuntu 18.04 lts) does the same as 0.9.4 on Windows: does the update but does not successfully load the new parts. Looking closer, the problem appears to be the upgrade is loading something other than the current fritzing-parts repo from https://github.com/fritzing/fritzing-parts.git as it has for instance AMT103_1.fzp (one of the frc parts) but whatever version was loaded by the parts update does not. As far as I know the updated should be loading from the above parts repository url but it doesn't appear to be.

KjellMorgenstern, 2019-08-30:
Expected is
0.9.3beta : should use master and upgrade flawless, only Calliope and others, but not yet FRC parts

0.9.4beta Win + Mac: same as 0.9.3beta, uses master branch of Fritzing-parts

0.9.4beta Linux only: Uses develop branch of Fritzing-parts, Calliope and FRC parts should be available

vanepp, 2019-08-30:
Ah! As usual github is baffling me. It looked to me like like the update should be in master, because when I cloned what I think is fritzing-parts master (but come to think of it is probably really develop) the parts were there. As long as this as expected I'll close this as a false alarm.

closed 2019-08-30 by vanepp