fritzing/fritzing-parts #219: PCF8574 in core parts.

vanepp, 2019-09-14:
Someone in the forums had a problem with a fzz file. While I think the most likely answer is the fzz file got corrupted, it would be prudent to check the git logs to make sure a change didn't happen in the PCF8574 part in core. The details of the problem can be found here in the forums:

http://forum.fritzing.org/t/cannot-find-part-in-old-project/7371/7

but essentially the moduleId in the fzz file is 'PCF8574' but the actual moduleId in current core parts is 'PCF8574A_f92705dc5bbcb278fb8f55ff08c7872a_1'. If that changed in core parts at some point (the part looks to have been added some time in 2018), which I don't know enough about searching github to figure out, then the original part needs to go in to obsolete (which it doesn't appear to have currently) with the necessary changes  need to be made to update it to the new value (which I again don't know how to make) so fzz files like this one will update to the new part correctly. I think the most likely answer is something unknown truncated the moduleId in the user's fzz file but a check of the git logs seems prudent just in case there was an unnoticed change in core parts which will cause more errors like this for other folks.

KjellMorgenstern, 2019-09-14:
Indeed the module ID was changed like that in June 2018, and this change became active when I merged nightlyBuild / develop to master some weeks ago.

Together with the change in the ID, also new svg files were added.

I created PR #220 as a start for fixing this, but I am not familiar yet with how the obsolete part mechanism works.

KjellMorgenstern, 2019-09-14:
I checked the fix agains the original lcd1602_i2c.fzz file, and it now will find the part. But as I understand it, Fritzing should also notify the user that the part was replaced with a fixed version. Unfortunately, I could not yet find out how the obsoletion mechanism works.

vanepp, 2019-09-14:
Last I saw, the obsolete parts doc had TBD (presumably to be done) in it and nothing else. I've been meaning to reverse engineer an obsoleted part and document how to do it but haven't gotten there yet.

KjellMorgenstern, 2019-09-14:
In #158 user mMerlin did some research about it.