fritzing/fritzing-parts #224: Arduino MKR WiFi 1010 library part is incorrect

NameOfTheDragon, 2019-10-02:
## Current Behaviour
The included library part for Arduino MKR WiFi 1010 is completely incorrect. It has the correct number of pins, but the pinout does not match the device. The appearance is also not correct (it appears to be an Arduino Fio). The MKR WiFi 1010 has more connectors in different locations and a WiFi chip and antenna.
[image]

This is odd, because there is a correct looking part in a Fritzing schematic on the Arduino Project Hub at https://create.arduino.cc/projecthub/Arduino_Genuino/puzzlebox-with-mkr-wifi-1010-7a39c4?f=1

**Build:**
Version 0.9.3 (b5c895d327c44a3114e5fcc9d8260daf0cbb52806 2016-04-19) 64 [Qt 5.6.1]

**Operating System:**
<Windows 10>

vanepp, 2019-10-10:
That would probably be because there is not a Arduino MKR WiFi 1010 in core parts as far as I can see. That would mean that you have loaded a custom part for the MKR WiFi 1010 and it is configured incorrectly in either family or variant or both and thus selects the fio part (which is in core parts). What MKR WiFi 1010 part are you using? If you post a url for the part you have I will have a look at it with a view to correcting it. SteelGoose posted a MKR1000 part in the forums back in 2016 but google search for "fritzing part Arduino MKR WiFi 1010" or "fritzing part Arduino MKR 1010" doesn't find a MKR 1010 part (although obviously someone has one somewhere.) The part you are using should be in either your mine or temp parts bins and can be exported by right clicking on it and clicking export part which will write the part to an fzpz file. It will be easiest to post that to the Fritzing forums (which understand .fzpz files), to post it here you need to add a .zip to the end of the file name. 

Peter

vanepp, 2019-10-31:
I expect we can close this one. This isn't a Fritzing error, but rather a user supplied part error, and I can't find the referenced part to suggest a fix to it. 

Peter

KjellMorgenstern, 2019-10-31:
Thanks Peter

closed 2019-10-31 by KjellMorgenstern