fritzing/fritzing-parts #227: Pin label offset on generic IC

thomasw-mitutoyo-ctl, 2019-11-08:
I used the generic IC to create a SN74LS74AN. Unfortunately, in the schematics view, the labels have an offset of roughly 1 pin, which is misleading. I tried to move the labels, but it's not possible.

[image]

Using Fritzing Version 0.9.3 (b5c895d327c44a3114e5fcc9d8260daf0cbb52806 2016-04-19) 64 [Qt 5.6.1]

vanepp, 2020-01-19:
How did you edit the pin labels (and on which operating system)? While this is 0.9.4, I don't remember 0.9.3b being different (and this is Win7, edited via Inspector in the lower right window via edit pin labels):

[image]

which appears to work correctly. That said I would normally edit the underlying svg file and that is much easier to screw up ... If you move the part so it snaps to the grid do the labels then line up (sometimes initial placement is not on the grid and that may cause other problems too?) If you post the .fzpz file (you will need to rename it to .zip to post it here, or post it in the Fritzing Forums) 
l will have a look at the part (although that will only tell us what is wrong with the part not how it got that way.)

Peter

thomasw-mitutoyo-ctl, 2020-01-20:
Sorry, I don't have Fritzing installed on my new machine yet. Fritzing wants 8 € *per download* (which is silly), which I'm not going to pay. You can close this issue. It seems I'm no longer interested in Fritzing and contributing parts.

KjellMorgenstern, 2020-01-20:
@thomasw-mitutoyo-ctl Fritzing tried various ways of funding, which where kind of successful at the time, but for the last years the project was stalling. Pushing users towards the payment, and clearly communicating that we expect a contribution seems to work. I admit that the current form is clumsy, but its the best fit I found for all the constraints.You can suggest improvements here: https://github.com/fritzing/fritzing-app/issues/3548

closed 2020-01-20 by KjellMorgenstern

DarkLiKally, 2020-02-02:
@KjellMorgenstern I've the same issue with the latest version of fritzing.
[image]

Steps how it appeared for me: 
1. Create generic IC
2. Adjust the pin labels (Edit pin Labels)
3. designed some circuit
4. went into PCB view and moved stuff around and routet some lines
5. deleted the IC by accident (wanted to delete a routing)
6. pressed strg + z
7. The labels went offset, also labels were reset 
8. Set labels again
9. I've the result seen in the screenshot above