fritzing/fritzing-parts #229: Internal connections problem in the creation of a part V: 0.9.4

fgcoca, 2019-12-16:
Creation of an Arduino UNO type part from the existing component and based on its editing.

- **Explanation of the need:**

The new component has male connectors for each digital or analog pin that connect directly to the corresponding female pin and next to the signal pin are also pins, also male, with Vcc and GND. This pin arrangement greatly facilitates the connection of a multitude of sensors and actuators without using a protoboard.

- **Problem encountered:**
[KS0172 UNO with Pin Header Interface.zip](https://github.com/fritzing/fritzing-parts/files/3967330/KS0172.UNO.with.Pin.Header.Interface.zip)

The problem described requires a good number of internal connections to make each pin connected to its corresponding place and what happens is that these connections are not made correctly. Apparently in protoboard they are defined but when we change to scheme or PCB we observe that said internal connections are made to the center of the component and not to the appropriate pin and when we try to make connections of the male pins to, for example, a protoboard both the schematic and the PCB they turn red and the connections are not correct.
Attached model created for a UNO type plate of Keyestudio.
Thanks and kind regards

KjellMorgenstern, 2019-12-16:
Part creators seem to run into this issue often.
TODO
1. Provide workaround
2. Create Fritzing feature to warn about this situation or just accept this format.

vanepp, 2020-01-19:
This is a case of operator error, not a bug. Fritzing is correctly indicating via the red rectangles that the part is invalid. The work around is (unfortunately because it is hard and takes a lot of time to learn) to make correctly formatted parts. It also has nothing to do with the bus layout (aka internal connections), but rather missing connectors in the svgs. There is an invalid null character in the fzp xml file which cause FritzingCheckPart.py to fail on an xml parse error, with that removed FritzingCheckPart.py reports the following errors, not all of them listed here, as there are around 20 screens or so of errors and not all which will cause a failure, but all of which are incorrect:

Error 47: File
'part.KS0172_325c9b4ccb00dad556880e5a20ce4622_25.fzp.bak'
At line 147

LayerId unknown doesn't match View breadboardView layerId breadboardbreadboard

This should be breadboard not breadboardbreadboard (breadboardbreadboard is a special layerId only used for breadboards because it trips special code to deal with breadboards) and breadboard not unknown. There are a lot of these but they may not be immediately fatal, but are as noted incorrect and likely to affect svg export.

These are what is causing the red rectangles due to pins defined in the fzp file but not in the associated svg files (and again there are a lot of them, 147 according to parts editor) and are the main problem:

Error 18: File
'part.KS0172_325c9b4ccb00dad556880e5a20ce4622_25.fzp.bak'

Connector connector45pin is in the fzp file but not the svg file. (typo?)

svg svg.breadboard.KS0172_380315f511fc81bf0d19dc08985407e4_1_breadboard.svg.bak

Error 18: File
'part.KS0172_325c9b4ccb00dad556880e5a20ce4622_25.fzp.bak'

Connector connector0pin is in the fzp file but not the svg file. (typo?)

svg svg.schematic.KS0172_380315f511fc81bf0d19dc08985407e4_1_schematic.svg.bak

Error 18: File
'part.KS0172_325c9b4ccb00dad556880e5a20ce4622_25.fzp.bak'

Connector connector1pin is in the fzp file but not the svg file. (typo?)

svg svg.schematic.KS0172_380315f511fc81bf0d19dc08985407e4_1_schematic.svg.bak

Error 18: File
'part.KS0172_325c9b4ccb00dad556880e5a20ce4622_25.fzp.bak'

Connector connector2pin is in the fzp file but not the svg file. (typo?)

Parts editor (which I rarely use) reports 147 missing connectors:

[image]

which causes the red rectangles and connection termination at the center of the part (as there is no where else for it to go.) I expect (if I wasn't too lazy to count them) FritzingCheckPart would agree with the count of 147. From the comment about prototype area, this may have been caused by removing the hybrid="yes" definition from pins on a prototype area without then assigning pins in the affected svgs (although that won't usually explain breadboard which is normally the only view with prototype connections active.) I can if desired fix this part up and point out the problems I find. 

Peter

fgcoca, 2020-01-19:
Thank you very much Peter for your extensive response. As soon as I can I
will return to the topic and try to deepen the theme of fritzing part
design, which by the way, I think is a great software.
Best regards

*------------------------------*
*Federico Coca*
*Maker, open source, 3D, Friki, FPGAWars*
*------------------------------*
Organizador de la feria de micro robótica e innovación GranaBot
@fgcoca

*e-NABLE volunteer*
 *------------------------------*
*+34 658914135*

vanepp, 2020-01-22:
Because this is a complex part and getting it correct isn't easy, I have made a new part with most of the corrections applied. It isn't entirely correct yet because of things I don't know: the ICSP and ICSP2 headers are labelled (and thus connected to) 5V, it is unclear if they should be VCC which switches between 5V and 3.3V or should always be 5V. If they should change to VCC, then the bus connections need to change to match. The major place changes are needed is in pcb, as many of the pads were missing and breadboard seems to be slightly the wrong scale (when compared to the standard Uno pcb footprint) so you will need to move the pads in the pcb svg around, as they are not currently in the correct place, only somewhat close, until they match the real board. The other thing to check would be the bus connections (aka internal connections) are correct. If you click on a pin (usually breadboard is the easiest), all pins that are connected to it should light up yellow. I think they are currently correct (I added all the servo connections which were missing) but there may be connections that should be there that are not. Your original problem was that breadboard was correct (except for one connector missing its label), but schematic and pcb were still the standard UNO missing all the servo connectors and some of the other extra connectors. They have all been added in this part (just remove the trailing.zip to leave the .fzpz file for the part as github won't accept .fzpz files):

[KS0172 UNO with Pin Header Interface.fzpz.zip](https://github.com/fritzing/fritzing-parts/files/4100191/KS0172.UNO.with.Pin.Header.Interface.fzpz.zip)

Peter

fgcoca, 2020-01-23:
Helo Peter,
First of all, thank you so much for such an excellent job and the patience you are having with me.
Secondly, to indicate that indeed the topic of 3.3 and 5V was not considered, at least for now, because only the level of the supply voltage on the pins changes.
The issue of the scale of the plaque, because the truth, I had not realized, but I will take it into account and correct for future creations.
Now I have to start reviewing everything and learn how to make models correctly with everything learned from these valuable comments.
Thank you very much Fritzing

closed 2020-01-23 by fgcoca

vanepp, 2020-01-23:
You are most welcome. The 8 tutorials here detail how this was done and may help you:

https://forum.fritzing.org/t/part-creation-howto-part-1-breadboard-and-pcb/7692

Peter