fritzing/fritzing-parts #236: Pin label offset on generic IC

DarkLiKally, 2020-02-14:
Hello,
I wanted to pick up the issue #227 again because I've more information about the error posted there after it was closed and the issue is still a problem for other users. There was sadly no reaction there, so I decided to create a new issue.

Please read the above mentioned issue first. Here are my additional informations:

I've the same issue with the latest version of fritzing.
[image]

Steps how it appeared for me: 
1. Create generic IC
2. Adjust the pin labels (Edit pin Labels)
3. designed some circuit
4. went into PCB view and moved stuff around and routet some lines
5. deleted the IC by accident (wanted to delete a routing)
6. pressed strg + z
7. The labels went offset, also labels were reset 
8. Set labels again
9. I've the result seen in the screenshot above

UPDATE:

I've fixed the problem for me by opening the project file with an archive editor and extracting the xml file (fz-file).

I've edited the geometry part of the part for schematic view. I have a second copy of the IC on my project so I compared the two, because the second one was still working. I've noticed that the attributes m31 and m32 were different for the two parts. I've copied over the values from the working one, saved and repacked the archive. The component was slightly moved to the side when reopening the projct in fritzing, but no problem, the Labels were placed correctly now. I've dragged the component back into it's desired place and everything looks fine now.

                    <geometry z="2.50364" x="157.5" y="103.5">
                        <transform m11="0" m12="1" m13="0" m21="-1" m22="0" m23="0" m31="112.5" m32="-31.5" m33="1"/>
                    </geometry>

Hope this information helps you to find and resolve the issue.

vanepp, 2020-02-15:
I think this is likely covered by fritzing-app issue #3610 which is active when someone gets time to work on it. It appears Fritzing is substituting an internal svg for the one you supply, so one work around  that I used on one case of this, is to modify the schematic svg. The part I did this on is available here:

https://forum.fritzing.org/t/ic-component-creation/8332

If you unzip the .fzpz file to extract the svgs you will see that schematic svg is incomplete (because I only changed 1 pin to get the spacing correct to match the substitution, and left the others incorrect) but the part itself is complete (because the svg was substituted. There is discussion in yet another thread about a way this can likely be fixed by making a code change and requiring labels in the svg for the pin labels, but that needs a code change which is on my list but no where near the top of it.

mMerlin, 2020-02-15:
@vanepp You can include a pointer directly to the fritzing app issue by including a reference link here.  Either as you did for the forum post, by just pasting in the link or by using the "add a link" post editing button.  I am used to markdown syntax, so I normally just type the brackets in.

```
[issue 3610](https://github.com/fritzing/fritzing-app/issues/3610)
```

Like this: frtizing-app [issue 3610](https://github.com/fritzing/fritzing-app/issues/3610)

Makes it easier for the OP to check if that seems to cover their case.  Or potentially add more detail to the other issue.

failiz, 2022-01-02:
I cannot replicate the offset in 0.9.9, but the labels are reset which is a bug. However, the generic IC is created dynamically in Fritzing. Thus, this bug should be moved to the app repository as it is not related to the parts in the parts repository.