fritzing/fritzing-parts #239: Providing a component for Beitian BN-220 GPS

albfan, 2020-03-01:
Using linux I was able to create this component. If it is interesting I would like to include into fritzing parts.

[Beitian BN-220 GPS.zip](https://github.com/fritzing/fritzing-parts/files/4272065/Beitian.BN-220.GPS.zip)

I did it with inkscape. If it's possible (maybe there's other way) I would like some review (as I'm sure it can be improved.

About creating parts with inkscape I can provide some feedback if needed so it's easy to create fritzing parts with inkscape (the font size seems to be problematic)

vanepp, 2020-03-12:
My first suggestion is be lazy. In this case that means do a google search for "fritzing part bn-220 gps" which turns up this version I made for someone a year or so ago:

https://forum.fritzing.org/t/i-look-for-the-gy-86-sensor-and-the-bn-220-gps-modul/7603

which I will eventually put in to core parts. That said there are a number of things wrong in your part.
It doesn't appear to be life size in breadboard and the pins don't have a terminalId (neither does mine, but the terminal is a square in the 1.27mm connector so it doesn't need one) which is why the wires in breadboard are attaching in the middle of the pin. For you configuration a square moved to the bottom end of the pin would cause the wires to connect correctly. 

[image]

Schematic has the same problem plus the pins are not aligned in .1in boundaries. Note in my part there is a  10 by 10 square with id set to connectorxterminal on the right end of each pin which is why the wires connect correctly. This image has the pins connect at 45 degrees to make sure the terminalId is correct:

[image]

and this one moved the connector so it should match the pins exactly (if they were on .1in boundaries) to demonstrate the misalignment:

[image]

Note the pins on your part are slightly narrower than .1in spaced. 

PCB is missing the copper1 group and thus only connects on the bottom or the board not the top.

[image]

Note that on my part every other pin is routed on the top of the board (the lighter colored trace), on your part because it only has copper0, the traces will only route on the bottom of the board which is incorrect for a through hole part (and a SMD part would only have copper1 not copper0.)

@albfan  "NOTE1: Is it possible to include a front and rear view in protoboard? for this component seems interesting"

Fritzing does not support bottom of the board in breadboard view (and changing that seems likely to be difficult.)

@albfan " What is the best way to design a component that is wired with a connector?"

Any way that works well is fine. In my case I chose to leave the connectors at 1.27mm in breadboard (the more usual practice is to move them to .1 centers so that wires align.) In PCB I used a 4pin .1in header (I could have also used a 1.27mm connector) because wires can be soldered in to the board to connect to the connector on the gps. 

There are tutorials on making parts available in the Fritzing forums (and a number of us are happy to help people make parts) here:

http://forum.fritzing.org/t/fritzing-part-creation-video-tutorial-series/2647/3

https://forum.fritzing.org/t/part-creation-howto-part-1-breadboard-and-pcb/7692

albfan, 2020-03-12:
Sure, this was more about doing a review and knowing the workflow. Thanks for looking at it

Let's then close this

closed 2020-03-12 by albfan