fritzing/fritzing-parts #252: 78xx series regulator in TO-92 package

davidperrenoud, 2020-03-30:
_From [eero.ha...@gmail.com](https://code.google.com/u/109634745033994792479/) on December 29, 2013 03:47:46_

What steps will reproduce the problem? 1. Searching 7805 regulator ->
- you get a 5 volt regulators OK
  1. Select the to-92 package
- in Inspector the Tag says reg_78xx OK 3. Attach your sketch file and/or custom part files to the bug report. - What is the expected output? What do you see instead? - in Breadboard window it says it is 78xx - OK
  - in schematic window the component too small compared to grid
  - in PCB window it has wrong legs in wrong order
  - in Inspector the Properties say it is 78005 not 7805
  - in Inspector you can select Voltages 1V, 1.25 -36V, 3.3V, 5V and Adjustable but you can not select other 78xx series voltages 6V, 7V, 8V, 12V and 15V ( 78xx series regulators are not adjustable) What version of Fritzing are you using? On what operating system? - Version 0.8.5 (b8b2c 2013-12-16) [Qt 4.8.5]
  - in Linux Please provide any additional information below. - I have not checked the other packages 

Thank You 
- Eero -

_Original issue: http://code.google.com/p/fritzing/issues/detail?id=2842_

KjellMorgenstern, 2020-03-30:
According to
http://users.ece.utexas.edu/~valvano/Datasheets/LM78L05.pdf

I think the correct order would be IN , GND , OUT for the PCB "view from above".
Current part sparkfun-poweric_to-92_pcb.svg connects in the order GND, IN , OUT.

KjellMorgenstern, 2020-05-02:
When fixing the 7805 connection, make sure to test the obsoletion mechanism. Swapped connection should be removed, so the user has to decide what the proper fix is for the circuit. If this does not work, the broken 7805 should be removed, and replaved with a fixed, but without using the obsoletion mechanism.

mMerlin, 2020-05-03:
Actually I think you meant that if the standard obsolete mechanism does not work to replace the bad 7805 with the good one, then the bad should be obsoleted, but WITHOUT using the replacedby attribute. So it goes to obsolete, but Fritzing will not attempt to replace it in existing sketches, but new sketches will find the good one in core.