fritzing/fritzing-parts #253: Raspberry PI3 pcb is slightly incorrect.

vanepp, 2020-04-22:
While making a Pi 4B part from the current PI3 part it appears that the GPIO connector in the current 3 part (and maybe the other PIs, I haven't checked yet) doesn't match the RPI mechanical drawing available here : 

https://www.raspberrypi.org/documentation/hardware/raspberrypi/mechanical/rpi_MECH_3b_1p2.pdf

(the center line of the GPIO header is a little high.) I corrected that for the 4B part and will eventually redo the PI3. However in case eventually never comes or someone else wants to do it, I'll document the issue here. The center line of the GPIO pins should be the same as the center line of the mounting holes, as we see here, it is not:

[image]

There are a number of other issues with this (and the other RPI parts) that should be fixed as part of a general cleanup of core parts. 

Peter

KjellMorgenstern, 2020-11-11:
@vanepp Did you work on this? Otherwise I'd look into this next.

vanepp, 2020-11-11:
No, I haven't gotten to it yet so feel free.

Peter