fritzing/fritzing-parts #254: Several pcb svg files seem to be missing

mMerlin, 2020-06-06:
I happened to be looking for the pcb image for a core part, and could not find it. A bit of research seems to show that several pcb files are missing, and/or the wrong file is being referenced from the part definition.

```sh
% grep "pcb/jumper_" /path/to/data_files/fritzing-parts/core/*
/path/to/fritzing-parts/core/Antenna.fzp:            <layers image="pcb/jumper_1_100mil_pcb.svg">
/path/to/fritzing-parts/core/Barometric Pressure Sensor.fzp:            <layers image="pcb/jumper_7_100mil_pcb.svg">
/path/to/fritzing-parts/core/Basic Force Sensing Resistor (FSR).fzp:            <layers image="pcb/jumper_2_100mil_pcb.svg">
/path/to/fritzing-parts/core/basic_fsr.fzp:   <layers image="pcb/jumper_2_100mil_pcb.svg">
/path/to/fritzing-parts/core/basic-toggle-switch.fzp:   <layers image="pcb/jumper_3_100mil_pcb.svg">
/path/to/fritzing-parts/core/battery-AA.fzp:            <layers image="pcb/jumper_2_100mil_pcb.svg">
/path/to/fritzing-parts/core/Battery block 9V.fzp:            <layers image="pcb/jumper_2_100mil_pcb.svg">
/path/to/fritzing-parts/core/dc_motor.fzp:            <layers image="pcb/jumper_2_100mil_pcb.svg">
/path/to/fritzing-parts/core/generic_sip_2_300mil.fzp:   <layers image="pcb/jumper_2_100mil_pcb.svg">
/path/to/fritzing-parts/core/infrared_proximity_sensor.fzp:            <layers image="pcb/jumper_3_100mil_pcb.svg">
/path/to/fritzing-parts/core/loudspeaker.fzp:            <layers image="pcb/jumper_2_100mil_pcb.svg">
/path/to/fritzing-parts/core/microphone.fzp:            <layers image="pcb/jumper_2_100mil_pcb.svg">
/path/to/fritzing-parts/core/peltier element.fzp:            <layers image="pcb/jumper_2_100mil_pcb.svg">
/path/to/fritzing-parts/core/piezo sensor.fzp:            <layers image="pcb/jumper_2_100mil_pcb.svg">
/path/to/fritzing-parts/core/pot_big.fzp:   <layers image="pcb/jumper_3_200mil_pcb.svg">
/path/to/fritzing-parts/core/servo.fzp:   <layers image="pcb/jumper_3_100mil_pcb.svg">
/path/to/fritzing-parts/core/solenoid.fzp:            <layers image="pcb/jumper_2_100mil_pcb.svg">
/path/to/fritzing-parts/core/stepper_motor_bipolar.fzp:            <layers image="pcb/jumper_4_100mil_pcb.svg">
/path/to/fritzing-parts/core/stepper_motor_unipolar.fzp:            <layers image="pcb/jumper_6_100mil_pcb.svg">
/path/to/fritzing-parts/core/Textile Potentiometer.fzp:   <layers image="pcb/jumper_3_100mil.svg">
/path/to/fritzing-parts/core/Tilt switch.fzp:            <layers image="pcb/jumper_2_100mil_pcb.svg">
% ls -1 /path/to/fritzing-parts/svg/core/pcb/jumper_*
/path/to/data_files/fritzing-parts/svg/core/pcb/path/to/.svg
/path/to/fritzing-parts/svg/core/pcb/jumper_3_100mil.svg
% git log
commit 023d0e492328b15a33a9b32e9aa871bcfdbfd010 …
```
From that, only a single part (Textile Potentiometer) matches the existing pcb svg file.

There **ARE** more jumper files in obsolete, though not the names seen in the current parts. The count of missing versus obsolete matches. Did some cleanup process (like a (long ago) rename to get better file names) happen that moved files to obsolete, instead of keeping a copy with a new name?

```sh
% ls -1 /path/to/fritzing-parts/svg/obsolete/pcb/jum*
/path/to/fritzing-parts/svg/obsolete/pcb/jumper10.svg
/path/to/fritzing-parts/svg/obsolete/pcb/jumper11.svg
/path/to/fritzing-parts/svg/obsolete/pcb/jumper12.svg
/path/to/fritzing-parts/svg/obsolete/pcb/jumper14.svg
/path/to/fritzing-parts/svg/obsolete/pcb/jumper16.svg
/path/to/fritzing-parts/svg/obsolete/pcb/jumper18.svg
/path/to/fritzing-parts/svg/obsolete/pcb/jumper1.svg
/path/to/fritzing-parts/svg/obsolete/pcb/jumper20.svg
/path/to/fritzing-parts/svg/obsolete/pcb/jumper2-3_5mm.svg
/path/to/fritzing-parts/svg/obsolete/pcb/jumper2.svg
/path/to/fritzing-parts/svg/obsolete/pcb/jumper3-200mil.svg
/path/to/fritzing-parts/svg/obsolete/pcb/jumper3-3_5mm.svg
/path/to/fritzing-parts/svg/obsolete/pcb/jumper3.svg
/path/to/fritzing-parts/svg/obsolete/pcb/jumper4.svg
/path/to/fritzing-parts/svg/obsolete/pcb/jumper5.svg
/path/to/fritzing-parts/svg/obsolete/pcb/jumper6.svg
/path/to/fritzing-parts/svg/obsolete/pcb/jumper7.svg
/path/to/fritzing-parts/svg/obsolete/pcb/jumper8.svg
/path/to/fritzing-parts/svg/obsolete/pcb/jumper9.svg
```

`git log --oneline «sample file names»` reports `12f12d6c inital commit - reset parts history` as the only commit that references those part and svg files. Even the one that matches. That commit is from Wed Apr 20 12:54:36 2016 +0200, and is confirmed to be the oldest commit in the current repository.

KjellMorgenstern, 2020-10-27:
The jumper pcb graphics are generated by Fritzing on the fly.

closed 2020-10-27 by KjellMorgenstern

mMerlin, 2020-10-27:
Does that mean that any part (core or custom) that needs the basic 'jumper' pcb footprint can just use a special name for the pcb view? `pcb/jumper_«number of pins»_100mil_pcb.svg`, or even (also) replace the `100` with the desired pin spacing?

Another thing to get into the documentation.

What other images are generated on the fly, that can be used when creating addition parts?