fritzing/fritzing-parts #257: Ultrasonic sensor pin label is interchanged

Ajith-stark, 2020-07-28:
Ultrasonic sensor schematic pin "labels" are interchanged with GND pin and ECHO pin.
[image]
You can see in the image when I clicked on ECHO pin its highlighting pin 4 which labeled as  GND.
its problem only with Schematic diagram 

Version 0.9.4 
(CD-498-0-a1ffcea 2019-12-01) 64 [Qt 5.9.5]

LInux

search "HC-SR04"
There are two ultrasonic sensors and bug is in sensor which named as "HC_SR04 Ultrasonic Distance Sensor" (showed in the inspector tab)

HaraldRau, 2020-11-05:
Hello, i have changed GND and ECHO, here are the zip-file:

[HC-SR04 Ultrasonic Distance Sensor.fzpz.zip](https://github.com/fritzing/fritzing-parts/files/5492826/HC-SR04.Ultrasonic.Distance.Sensor.fzpz.zip)

The result for search "HC-SR04" hast tow parts. The second part is a bad double from this part. You can delete the double.

Must i open a new issue for delete this part?

KjellMorgenstern, 2020-11-05:
Hello Harald, thanks for the fixes.
I ran the parts obsoletion. That will show a hint about the potential issue for existing circuits. Both of the hc-sr04 are replaced with a part hs-sr04_bf8299a_002.fzp

I did not yet verify that the application can deal with two different parts replaced by one single part, but we are still on develop branch, and can simply remove the bad double without replacements if it causes issues.

Can you apply the fix on the sr04_bf8299a_002 schematic?
See PR #276 

Note: Some version of the FritzingCheckPart script do write operations on the file, which is not ideal for reviews and version management. When you use it, please make sure to not apply unintended changes.

HaraldRau, 2020-11-05:
... all right.
I'm trying to fix PR #276.

Note: I will never use the FritzingCheckPart script.

HaraldRau, 2020-11-05:
In 'hs-sr04_bf8299a_002.fzp' does the connection seem to be correct, when I examining the correct file?

`<connector id="connector2" name="ECHO" type="male">
   <description>ECHO</description>`

` <connector id="connector3" name="GND" type="male">
   <description>GND</description>`

[Siehe: Obsolete core/hc-sr04_ultrafix.fzp](https://github.com/fritzing/fritzing-parts/commit/16942b1c65d117e0115535218105f105de4770d5#diff-767599940336b74e819e268f04554ab1b23ad3f067741f3a30071823dfdba6fe)

KjellMorgenstern, 2020-11-05:
Maybe the issue is in the schematics svg file, attaching the connector2 to the GND element.

HaraldRau, 2020-11-05:
I hope i find in Line 41-44 the problem of hs-sr04_bf8299a_002.svg:

line 41 connector2pin -> connector3pin
line 42 connector2terminal -> connector3terminal
line 43 connector3pin -> connector2pin
line 42 connector3terminal -> connector2terminal

But i can't edit the file. Here is the copy of  hs-sr04_bf8299a_002.svg with changed pin.
[hs-sr04_bf8299a_002.svg.zip](https://github.com/fritzing/fritzing-parts/files/5493825/hs-sr04_bf8299a_002.svg.zip)

Sorry for the changed line numbers and the additional entries. They're from Inkscape