fritzing/fritzing-parts #260: JQC-3F(T73) Relay - new item

davidperrenoud, 2020-10-14:
_From [fisc...@gmail.com](https://code.google.com/u/113716745659222267662/) on April 20, 2014 19:52:03_

Sorry, I couldn't find the new item form. Is it possible to add this Relay to Fritzing? It is very generic, comes in 12v, 5V and 3V versions and works very well with Arduinos and RPis. It's almost the standard relay being sold on Amazon. 

Let me know what info you need.

Thanks

Helmut

_Original issue: http://code.google.com/p/fritzing/issues/detail?id=2948_

KjellMorgenstern, 2020-10-14:
Looks very similar to the D1 Mini which we already have.

mMerlin, 2020-10-15:
> I couldn't find the new item form

"forum", not "form". Click "Forum" on [Fritzing site](https://forum.fritzing.org/) home page. I expect you will need to create an account and login to be able to post.

I created a FAQ entry recently for [Where can I find "specific part"](https://github.com/fritzing/fritzing-parts/wiki/FAQ:Where-can-I-find-%22specific-part%22).