fritzing/fritzing-parts #268: LED marks for anode and cathode in PCB view?

KjellMorgenstern, 2020-10-27:
The markings for the generic and the 10mm LEDs in Fritzing differ. They don't have a datasheet attached that could be followed.
Therefore, I think they should at least use the same marking system.
Another solution would be to offer variants with differnt copper / silkscreen.
 
The 5mm LED:
Anode: Round pad, at flat side
Cathode: Square pad, at round side 

The 10mm LED (sparkfun)
Anode: Square pad, at flat side
Cathode: Round pad, at round side

According to
https://electronics.stackexchange.com/questions/210753/whats-the-standard-for-denoting-the-orientation-of-an-led-on-a-pcb
Anode: Round pad, round side
Cathode: Square pad, flat side

The electrolytic capacitor
Anode: Round pad, flat side, + sign
Cathode: Square pad, round side, no sign

LED6 is the sparkfun 10mm 
LED5 is the 5mm core part
[image]
[image]

cjdg, 2020-10-27:
why dont use the common markings for capacitors?

KjellMorgenstern, 2020-10-27:
@cjdg Can you refer to standard or widely used document?
Similar to the LEDs, Fritzing is have various capacitors. It party depends on the manufacturers specification, but where possible we should of course align them:

[image]

cjdg, 2020-10-27:
https://www.tips.modularparts.net/led-polarity-orientation-pcb/ "mostly" everybody agrees with this