fritzing/fritzing-parts #284: uno mounting holes?

davidperrenoud, 2020-11-05:
_From [irasc...@gmail.com](https://code.google.com/u/104729248032245122687/) on June 07, 2012 11:03:40_

http://fritzing.org/forum/thread/950

_Original issue: http://code.google.com/p/fritzing/issues/detail?id=2018_

KjellMorgenstern, 2020-11-11:
See PR #288

mMerlin, 2020-11-11:
Do we really want the mounting holes as a permanent part of the uno part? If an uno is mounted on a pcb (above or below, as shield or part), none, some, or all of the mounting holes might be wanted for the project / pcb. Extra holes can be added using the "Hole" part, but there is no way to remove them when they are not wanted. Making them optional would be good, but that would be (back to) a code change. At least AFAIK, the sub-parts are only available for schematic view.

KjellMorgenstern, 2020-11-12:
Yes exactly. The PR adds the holes to the silkscreen layer, but should not modify the copper or add a drill hole.
The user could then use the silkscreen as guide to manually add holes.

Do you think the silkscreen can be mistaken as drill hole?

PCB subparts do not exist yet, I created [issue 3746](https://github.com/fritzing/fritzing-app/issues/3746) to think about that feature.

mMerlin, 2020-11-12:
If it visually looks like a hole, somebody will manage to expect it to create a drill hole. There is no way to always have everyone understand the same thing.