fritzing/fritzing-parts #299: netduino and netduino plus do not completely align to grid in bb view

davidperrenoud, 2020-11-17:
_From [irasc...@gmail.com](https://code.google.com/u/104729248032245122687/) on September 20, 2011 21:39:17_

What steps will reproduce the problem? 1. 2. 3. Attach your sketch file and/or custom part files to the bug report. If the sketch uses custom parts, save the sketch as shareable (under the file menu). What is the expected output? What do you see instead? What version of Fritzing are you using? On what operating system? Please use labels and text to provide additional information.

_Original issue: http://code.google.com/p/fritzing/issues/detail?id=1657_

mMerlin, 2020-11-17:
The problem is that the header pins them selves are not aligned to a 0.1 inch grid. Like the Arduino UNO (and others), the header group containing "Aref" is offset from the header beside it by 0.16 inches, instead of 0.1 inch. That is the way the physical part itself is designed, so the Fritzing needs to match that. There is no way to get all of the pins to to align to the 0.1 inch grid, when the physical pins do not align. At least while keeping the bb view matching the physical part pin spacing.

KjellMorgenstern, 2020-11-17:
Thanks. So the challenge would be to design an Breadboard Adapter board, and package that to the part. The Netduino is discontinued and was replaced with the Meadow board. Closing.

closed 2020-11-17 by KjellMorgenstern

mMerlin, 2020-11-18:
The uno and other boards that use the same header spacing have the same problem. Even more so, when the ICSP headers are included. What I think is really needed, is a way to specify which connector in the part to use as the 'origin', to align to the grid. That way the part creator can choose which block(s) of pins align, and which don't. Like:

```xml
<connector id="connector9" origin="true>
```

or 

```xml
<connector id="connector9">
  <breadboardView>
    <p layer="breadboard" origin="true" …/>
```

These cases are all microcontroller boards, and are not normally placed on breadboards. They typically (should) have type="female" connectors, so that parts (shields) can connect to them. The alignment is still annoying, because wires do not connect "straight" to (some of) the pins. No really "good" fix, since want to keep the part (bb view) matching the physical board. It would be possible to create variant parts where the bb view connectors are deliberately not where they are on the physical board. That would look a bit cleaner when running wires directly to the pins. It would also break if attempting to position an "accurately" designed shield/hat board over top.

Discontinued parts may not get 'priority', but they can still be useful. They still exist 'in the wild'.