fritzing/fritzing-parts #306: Symbol of preset is wrong

bobb3, 2020-12-29:
Electronic circuit symbol of preset is wrong

failiz, 2021-12-27:
I made a new version of the potentiometer (schematic symbol), which is below. I would prefer to replace the symbol for a better one (the wiper should be straight instead of having a right angle). I see this as a temporral solution until we make a better one. If a better one is approved, we should (1) obsolete the old ones or (2) replace the potentiometers in the core to produce new symbols and leave the old ones in the schematics produced with old versions of Fritzing. 

[basic_poti.zip](https://github.com/fritzing/fritzing-parts/files/7782002/basic_poti.zip)
[image]

failiz, 2023-08-01:
This has been fixed in 1.0.0.