fritzing/fritzing-parts #318: core parts need cleaning

davidperrenoud, 2021-03-21:
_From [irasc...@gmail.com](https://code.google.com/u/104729248032245122687/) on September 09, 2009 15:00:26_

There are lots of core parts in need of cleaning.  
1.  Many still have sodipodi stuff; sometimes this has been cleaned up, but 
   there are still a dangling attributes left
2.  Every element in a part that will be visible should be inside a <g 
   id="[layer-id]"> element (call this the g-layer-id element).  In other 
   words, this <g> element should be either at the top level of the svg, or if 
   there are several layers, next level down.  Some parts don't have a g-
   layer-id-element.  Some parts don't have all visible subelements inside the 
   g-layer-id element.
3.  Before a part gets added to the core, one of us needs to check it out.  
   Dropping a candidate part into a sketch and exporting it as an svg seems to 
   be a good way to surface problems with a part.
4.  I think we should add new parts to some svn holding pen before they're 
   actually moved into core and released.

_Original issue: http://code.google.com/p/fritzing/issues/detail?id=812_

KjellMorgenstern, 2025-09-25:
This issue is open since 2009, but somewhat still valid. It should be split into some actionable sub-issues.

KjellMorgenstern, 2025-09-25:
Closing in favor of #424 .

closed 2025-09-25 by KjellMorgenstern