fritzing/fritzing-parts #345: Part Creation - Default to Bottom Layer?

IAmOrion, 2022-08-23:
When creating a new fritzing part, is it possible to set the part to default to the BOTTOM layer?
For example, Kailh Socket Mounts would normally be soldered to the underside of a PCB so it makes sense that by default when adding to PCB view it should default to bottom layer

KjellMorgenstern, 2022-08-24:
The user can select the active layer in the Fritzing UI. If SMD components would suddenly appear on a different layer, that would be quite surprising behavior.

[image]

So even if it can be done, I am not convinced that it should be done. 
Instead, maybe you can add some mounting instructions in the description area.

IAmOrion, 2022-08-24:
> The user can select the active layer in the Fritzing UI. If SMD components would suddenly appear on a different layer, that would be quite surprising behavior.
>
> So even if it can be done, I am not [...]

I know how to arrange a component etc and place it on the bottom later.  But for certain usage cases (Eg, this socket for cherry mx switches - https://www.adafruit.com/product/4958) where the component is designed to be mounted on the underside/bottom of the PCB it would be more efficient to just drag the component from the parts bin and have it automatically on the bottom layer since it's 99% certain you wouldn't ever have this on a TOP layer

KjellMorgenstern, 2022-08-24:
Ok, you design it in revers. 
A user wants to mount a part on the bottom layer. So they will select the bottom layer. Then the users starts placing the parts. After a few parts, user realizes that all the parts have been placed on the top layer, because they were designed in reverse. User swears, and starts over. Eventually reports a bug.

Alternative: Parts are *always* placed on the bottom layer. Now, someone for sure will have the requirement to place it on top, we don't question their reason. But it is not possible. User swears and reports a bug.

This complicates things. The efficiency improvement you claim is literally one single click, but only if the user already exactly knows about then non standard behavior of the part.