fritzing/fritzing-parts #347: obsolete.py improvements

vanepp, 2022-09-11:
I'm learning and documenting obsoleting, and as part of that have been poking at obsolete.py. Before I get too far I figured I should explain what I want to change and see if there is agreement or if I am overlooking something. 

bug:

The dom returned by get_dom isn't the root, but rather     return dom.documentElement and thus 

 s = new_fzp_dom.toxml("UTF-8", False)

doesn't emit the  <?xml version='1.0' encoding='UTF-8'?> line (because the dom isn't the root) on the output fzp files (obsolete and core.) The above format also requires python 3.9 or later to add the standalone=no parameter (which most Fritzing fzps have.) Returning dom from get_dom and adding the .documentElement to all other dom references fixes that. I don't think that Fritzing cares, because UTF-8 is the default, but vim doesn't recognize the file as xml (and thus doesn't enable the xml checks.) I expect this shouldn't be controversial. 

Next is an enhancement which may be more controversial. I'd like to remove the revision argument and replace it with the version in the fzp file with the new fzp and the svgs getting the current version plus 1. There are at least a couple of files in core parts that have a version 1.1 currently, so they would need to be updated to be 1 (so the version is an int.)  If there is no version tag in the fzp file the code currently sets it to 2, and I think that 1 is a better choice (with the new fzp being 2.) In the case of a fzp with a version number the new version would be the current version + 1. I'm not sure if the revision argument is intended to allow something that I'm not aware of. I think (but haven't tested yet) that the current code will break if there is a version number in the fzp file. I think this code is going to break if there is a version tag in the fzp. 

    versions = obsolete_fzp_dom.getElementsByTagName("version")
    if not versions:
        doc = obsolete_fzp_dom.documentElement.ownerDocument
        version = doc.createElement("version")
        version.appendChild(doc.createTextNode("1"))
        before_node = obsolete_fzp_dom.documentElement.childNodes[1]
        obsolete_fzp_dom.documentElement.insertBefore(version, before_node)
        print("Set version to 1 as it didn\'t exist")
    else:
        version = versions[0]
        print("Set version to %s" % version)

    new_fzp_dom = deepcopy(obsolete_fzp_dom)

    version.setAttribute("replacedby", new_module_id)

the last version.setAttribute I think will fail when the version in the else above has been replaced with the string for the version number as it will no longer be a Element but rather a string (the code above is in a new order than the current obsolete.py so as the deepcopy will get the new version element when there wasn't a version tag in the original fzp.) It also moves the version to right after the module at the top of the fzp to match the rest of the Fritizng fzps. 

Next the new svg filenames will have the new version and the layerId appended so 

svg/core/breadboard/Dual_VNH2SP30_Motor_Driver_2_55dde35_002.svg

would become 

svg/core/breadboard/Dual_VNH2SP30_Motor_Driver_2_55dde35_2_breadboard.svg

where the _2 is the new version number (this part doesn't have a version tag) and the _breadboard is the layerId. The purpose of this is to make it easier to make fzpz files. The original version has the same file name for all the svgs and thus you can't copy them to a single directory to make a .fzpz file because they will overwrite each other. The new format tells you what you need to append (svg.breadboard. in this case) to make a fzpz compatible svg. The _002 in the original file name is the revision parameter and thus the question above if that is intended for something I''m not aware of. 
In the most common version in core parts (4) the new version would become 5.

Next I added code to search the bins/core.fzb file for the moduleId. I don't know what bins/core.fzb is for exactly (as all the core parts are not in it, but core/Dual_VNH2SP30_Motor_Driver.fzp is and if you don't replace it with the new moduleId you get the obsolete part in a core parts search. I expect we should possibly add all the core parts to this file as I don't see any reason why core/Dual_VNH2SP30_Motor_Driver.fzp is in it but most of the arduinos are not. It doesn't seem to be related to where in the parts bins the part appears. Nor does it appear to be parts that aren't in a bin (because the "arduino_Uno_Rev3(fix)" is there and it is in a bin as well!) There may be other files (that I haven't run across yet!) that also need updating. It is also unclear to me where you get the modelIndex field if you were adding a part (it doesn't apply in this case because I am assuming the current modelIndex will do, although that may not be true, it appears to work in the Dual_VNH2SP30_Motor_Driver.fzp case but may not in another case!) 

Last I intend on adding a //obsolete// to the obsolete.fzp family attribute to match all the other obsoleted parts. I gather that replacedby is the only required field but this is easy enough to add and is useful in the case of bins/core.fzb to tell you you are looking at the wrong instance of the part. 

Finally I don't think

 "TODO: Support obsoleting a part without having a replacement. This could
 be useful for parts whic are not available anymore."

is a good idea. I suggested it in the forums and it was pointed out that even though the part is no longer available for sale, they are in people's possession of it, and it is probably still available on ebay et. al. used, so, as long as the part is correct, it is likely still useful for old sketches if nothing else!

Comments? Corrections?

vanepp, 2022-09-14:
OK, I have what appears to he a working script with a variety of improvements (including being able to deal with the breadboard svg also being the icon svg!) Below are the test results first from the original script then from my new script. 

output from the original script:

$ obsolete.py core/Dual_VNH2SP30_Motor_Driver.fzp Dual_VNH2SP30_Motor_Driver

$ git status
On branch develop
Your branch is up to date with 'origin/develop'.

Changes to be committed:
  (use "git restore --staged <file>..." to unstage)
        new file:   core/Dual_VNH2SP30_Motor_Driver_44e6ab2_002.fzp
        renamed:    core/Dual_VNH2SP30_Motor_Driver.fzp -> obsolete/Dual_VNH2SP30_Motor_Driver.fzp
        renamed:    svg/core/breadboard/Dual_VNH2SP30_Motor_Driver_breadboard.svg -> svg/core/breadboard/Dual_VNH2SP30_Motor_Driver_44e6ab2_002.svg
        renamed:    svg/core/icon/Dual_VNH2SP30_Motor_Driver_icon.svg -> svg/core/icon/Dual_VNH2SP30_Motor_Driver_44e6ab2_002.svg
        renamed:    svg/core/pcb/Dual_VNH2SP30_Motor_Driver_pcb.svg -> svg/core/pcb/Dual_VNH2SP30_Motor_Driver_44e6ab2_002.svg
        renamed:    svg/core/schematic/Dual_VNH2SP30_Motor_Driver_schematic.svg -> svg/core/schematic/Dual_VNH2SP30_Motor_Driver_44e6ab2_002.svg
        new file:   svg/obsolete/breadboard/Dual_VNH2SP30_Motor_Driver_breadboard.svg
        new file:   svg/obsolete/icon/Dual_VNH2SP30_Motor_Driver_icon.svg
        new file:   svg/obsolete/pcb/Dual_VNH2SP30_Motor_Driver_pcb.svg
        new file:   svg/obsolete/schematic/Dual_VNH2SP30_Motor_Driver_schematic.svg

$ vi core/Dual_VNH2SP30_Motor_Driver_44e6ab2_002.fzp

missing first line of file

<module fritzingVersion="0.7.0b.02.01.5801" moduleId="DualVNH2f713248e766f4b528720e0f0e077c4d7">
    <author>David Thomasson</author>
    <title>Dual VNH2SP30 Motor Driver</title>
    <label>MD03A</label>
    <date>2012-02-05</date>
    <tags>

/version/

E486: Pattern not found: version 

no version tag in new .fzb file.

$ vi obsolete/Dual_VNH2SP30_Motor_Driver.fzp

missing first line of file.

<module fritzingVersion="0.7.0b.02.01.5801" moduleId="045cef49a96039c64e7363608e127209">
    <author>David Thomasson</author>
    <title>Dual VNH2SP30 Motor Driver</title>
    <label>MD03A</label>
    <date>2012-02-05</date>

no //obsolete// in family

    <properties>
        <property name="family">Motor Driver</property>

version at the bottom of the svg rather than after module as is usual with Fritzing parts. Version number is 2 not 1 as it should be (as this is the first version as it doesn't have a version tag in the original fzp)

        </connector>
    </connectors>
<version replacedby="DualVNH2f713248e766f4b528720e0f0e077c4d7">2</version></module>

$ obsolete.py core/Arduino-Mini-v5.fzp Arduino-Mini-v5
('git', 'mv', 'core/Arduino-Mini-v5.fzp', 'obsolete/Arduino-Mini-v5.fzp')
('cp', 'svg/core/breadboard/Arduino-Mini-v5_breadboard.svg', 'svg/core/breadboard/Arduino-Mini-v5_767ab6c_002.svg')
('git', 'mv', 'svg/core/breadboard/Arduino-Mini-v5_breadboard.svg', 'svg/obsolete/breadboard/Arduino-Mini-v5_breadboard.svg')
('git', 'add', 'svg/core/breadboard/Arduino-Mini-v5_767ab6c_002.svg')
set layer image to breadboard/Arduino-Mini-v5_767ab6c_002.svg
('cp', 'svg/core/schematic/Arduino-Mini-v5_schematic.svg', 'svg/core/schematic/Arduino-Mini-v5_767ab6c_002.svg')
('git', 'mv', 'svg/core/schematic/Arduino-Mini-v5_schematic.svg', 'svg/obsolete/schematic/Arduino-Mini-v5_schematic.svg')
('git', 'add', 'svg/core/schematic/Arduino-Mini-v5_767ab6c_002.svg')
[... 242 lines cut ...]
$ mv Arduino-Mini-v5_ba9176d_6.fzp part.Arduino-Mini-v5_ba9176d_6.fzp

owner@owner-PC /cygdrive/d/repos/fzpz
$ mv Arduino-Mini-v5_ba9176d_6_breadboard.svg svg.breadboard.Arduino-Mini-v5_ba9176d_6_breadboard.svg

owner@owner-PC /cygdrive/d/repos/fzpz
$ mv Arduino-Mini-v5_ba9176d_6_pcb.svg svg.pcb.Arduino-Mini-v5_ba9176d_6_pcb.svg

owner@owner-PC /cygdrive/d/repos/fzpz
$ mv Arduino-Mini-v5_ba9176d_6_schematic.svg svg.schematic.Arduino-Mini-v5_ba9176d_6_schematic.svg

owner@owner-PC /cygdrive/d/repos/fzpz
$ ls
part.Arduino-Mini-v5_ba9176d_6.fzp
svg.breadboard.Arduino-Mini-v5_ba9176d_6_breadboard.svg
svg.pcb.Arduino-Mini-v5_ba9176d_6_pcb.svg
svg.schematic.Arduino-Mini-v5_ba9176d_6_schematic.svg

and the .fzpz file ready to be zipped.

$ obsolete-fixed.py core/uM_FPUv2.fzp uM_FPUv2
('git', 'mv', 'core/uM_FPUv2.fzp', 'obsolete/uM_FPUv2.fzp')
Error: version must be only digits, not 1.1

The error message when the version field has something other tnan digits in it. We need to change the version on these 2 files from 1.1 to 1. This covers cases where there is no version tag, and where there is a version tag present, updates to bins/core.fzb and when the moduleId is not in bins/core.fzb, and when the version number is 1.1 (only 2 files I can see in core parts with that version number!) I'll see about making a pull request against obsolete.py in the parts repository.

KjellMorgenstern, 2023-01-26:
@vanepp I agree, we should not apply the obsoletion mechanism for discontinued parts. It might make more sense to just remove them from the bin, or add a "discontinued" property.

Regarding your commit fb6dd3c , you don't need to rename the script. Git works much better if push your modifications to the original file. Should I create such a modification based on your copy? Basically you just open the original file, make your modificaiton (in this case, copy your new version over it) and commit the change. I did not yet fully understand the bugs you fixed on the script.

vanepp, 2023-01-29:
"Should I create such a modification based on your copy? "

Sure if that is easy, if it is not I can just push the the change directly if that is easier.  Which parts of the change aren't clear and I'll try and explain why I did it as some of them may not be suitable. Most of them came from issues I found when trying to obsolete a couple of parts.