fritzing/fritzing-parts #356: Change default resistor label

KjellMorgenstern, 2023-01-25:
The "power" property of the label did not have a unit (it should be 0.25W instead of 0.25)
Since many circuits use the default of 0.25W, it makes sense to not show this in the label by default.

KjellMorgenstern, 2023-01-25:
Default label in schematic view:
[image]

KjellMorgenstern, 2023-01-25:
For existing circuits, the label visibilty is not changed, but the unit is now shown:
[image]

failiz, 2023-01-26:
This change was in one commit of my PR:
https://github.com/fritzing/fritzing-parts/pull/344/commits/eb8b1b1b98708d121c15f295f6bcca22a987409e

closed 2023-01-26 by KjellMorgenstern