fritzing/fritzing-parts #359: Trying to import fzpz files produces file not found error

emersonsc, 2023-02-13:
No matter what I do, i get an error similar to this one, saying a file in a specific folder doesnt exist. I've tried hand fulls of files. This is a fresh brand new install, ive never had Fritzing before.

[image]

nelsonrinconq, 2023-02-22:
[image]

KjellMorgenstern, 2023-02-22:
@nelsonrinconq This is not a file not found error. IDs must be unique, so you either have to change the ID or remove the previous version from Fritzing.

nelsonrinconq, 2023-02-22:
heavy search but ty, 
[image]

nelsonrinconq, 2023-02-22:
ty so much

KjellMorgenstern, 2023-02-22:
One tip: Fritzing will automatically allign the part on the first connector (usually connector0), so it is easy to fit on the breadboard. In your screenshot, the part doesn't look aligned, so there might be something wrong with connectors, maybe they are not in a "breaboard" layer group. And of course, the part needs to be rotated to automatically fit the breadboard.