fritzing/fritzing-parts #372: power supplies need to be oriented vertically

failiz, 2023-04-26:
See the battery power supply in the schematic view:
[image]
The battery appears rotated 90 degrees, which is not the common way of writing electronics (power at the top and ground at the bottom). This forces to rotate the part (which is annoying for advanced users) or contributes to creating "ugly" schematics by beginners.

The solution is simple, rotate the SVG file.

I tried to rotate it in the SVG file by just adding a `transform="rotate(90, 5.20345, 4.0132)"` see below:

```
<?xml version='1.0' encoding='UTF-8' standalone='no'?>
<!-- Created with Fritzing (http://www.fritzing.org/) -->
<svg xmlns:svg='http://www.w3.org/2000/svg' xmlns='http://www.w3.org/2000/svg' version='1.2' baseProfile='tiny' x='0in' y='0in' width='0.409722in' height='0.316in' viewBox='0 0 10.4069 8.0264' transform="rotate(90, 5.20345, 4.0132)">
<g id='schematic'>
<line class='other' x1='3.93347' y1='0.2032' x2='3.93347' y2='7.8232' stroke='#000000' stroke-width='0.4064' stroke-linecap='round'/>
<line class='other' x1='5.20347' y1='2.7432' x2='5.20347' y2='5.2832' stroke='#000000' stroke-width='0.4064' stroke-linecap='round'/>
<line class='other' x1='6.47347' y1='0.2032' x2='6.47347' y2='7.8232' stroke='#000000' stroke-width='0.4064' stroke-linecap='round'/>
<line class='other' x1='7.74347' y1='2.7432' x2='7.74347' y2='5.2832' stroke='#000000' stroke-width='0.4064' stroke-linecap='round'/>
<line class='other' x1='2.66347' y1='4.0132' x2='3.67947' y2='4.0132' stroke='#000000' stroke-width='0.1524' stroke-linecap='round'/>
<line class='pin' id='connector0pin' connectorname='-' x1='10.2835' y1='4.0132' x2='7.74347' y2='4.0132' stroke='#787878' stroke-width='0.246944' stroke-linecap='round'/>
<rect class='terminal' id='connector0terminal' x='10.2835' y='4.0132' width='0.0001' height='0.0001' stroke='none' stroke-width='0' fill='none'/>
<text class='text' font-family="'Droid Sans'" stroke='none' stroke-width='0' fill='#8c8c8c' font-size='0.881944' x='9.01347' y='3.64278' text-anchor='middle'>1</text>
<line class='pin' id='connector1pin' connectorname='+' x1='0.123472' y1='4.0132' x2='2.66347' y2='4.0132' stroke='#787878' stroke-width='0.246944' stroke-linecap='round'/>
<rect class='terminal' id='connector1terminal' x='0.123472' y='4.0132' width='0.0001' height='0.0001' stroke='none' stroke-width='0' fill='none'/>
<text class='text' font-family="'Droid Sans'" stroke='none' stroke-width='0' fill='#8c8c8c' font-size='0.881944' x='1.39347' y='3.64278' text-anchor='middle'>2</text>
</g>
</svg>
```
This works great on my computer, but Fritzing cannot draw it, see below message:
[image]
Any workarounds?

KjellMorgenstern, 2023-04-26:
I'd try to apply the transform attribute to the <g> element.

We would the move the current part to obsolete, and add new parts with corrected schematic orientation as default. Don't use a replaced-by so that existing projects will not have to change their schematic, especially if they already rotated the symbol.

failiz, 2023-04-26:
> I'd try to apply the transform attribute to the `<g>` element.
>
> We would the move the current part to obsolete, and add new parts with corrected schematic orientation as default. Don't use a [...]

The problem with that is that the shape of the part does not change, it keeps the original layout. I guess that we are going to replace it I can do it in Inkscape.

KjellMorgenstern, 2023-04-26:
Ah, 0.9.10 ignores the rotation if it is on the group element instead of the svg element ?

failiz, 2023-04-26:
No, I meant that the part is rotated, but the size of the part is still using the horizontal layout (but I did not try it in Fritzing). The size has to be transposed (the width is now the height) and the viewbox needs to be adjusted.

KjellMorgenstern, 2023-04-26:
Ah, yes. The same problem might apply for the Origin and therefore the grid alignment. And even if it would work in recent Fritzing builds, we should maintain some backward compatibility for these core parts. So we must flatten the svg graph after adding the transform.

vanepp, 2023-04-28:
At a request in the forums I created new (meeting the graphics and parts file format file standards) svgs for these three parts that I think fix all the listed issues. At least the battery  svg was dimensioned in px (which will cause scaling problems when edited with a current version of Inkscape due to the DPI difference since the original is 72DPI and current Inkscape uses 96DPI for px.)