fritzing/fritzing-parts #374: uno r4 wifi support

LYB030, 2023-06-30:
## Problem
Describes from a user perspective the current state and its shortcomings.

## Proposed Solution
Describes from a user perspective the ideal state of the implementation

KjellMorgenstern, 2023-07-10:
@roboteach-es We'll try to provide the vector graphic asap.

LisaChyhryna-fritzing, 2023-07-10:
@roboteach-es Hey! Here are the vector graphic for uno r4 wifi. 
[image]

LisaChyhryna-fritzing, 2023-07-10:
https://github.com/roboteach-es Hey! Here are the vector graphic for uno r4 wifi.
[image]

KjellMorgenstern, 2023-07-10:
Same file as above, but I have converted the white four circles into cutouts (so you will see through to the background) and ran 'scour' on it to reduce the size a bit.
[image]

There might be problems with the CSS selectors , not sure if Fritzing will respect those especially for SVG export, we should try. Qt can render them just fine.

mgesteiro, 2023-07-17:
> https://github.com/roboteach-es Hey! Here are the vector graphic for uno r4 wifi.

Thanks @LisaChyhryna-fritzing, I'm on it!

mgesteiro, 2023-07-17:
> There might be problems with the CSS selectors , not sure if Fritzing will respect those especially for SVG export, we should try. Qt can render them just fine.

I'll check this also, N.P.

KjellMorgenstern, 2023-07-24:
@vanepp Posted a version here: https://forum.fritzing.org/t/arduino-giga-r1-wifi/20039

failiz, 2023-07-26:
I would say that it would be great if we improve the schematic view of the parts too. Some problems to agree on and improve (pinging @vanepp as he usually makes a lot of parts):

* The current Arduino parts and the ICs parts in the core have connectors of 0.2 inches while the other simple parts (resistors, LEDs, pots, switches) have 0.1 inches connectors. I would say that all the parts should have 0.1 inches connectors.
* The connectors of all Arduino parts are not highlighted in red as disconnected as is usual in Fritzing. The turn green when connected with a wire though, see the picture below.
* The part could have some general improvements like the part that I think was created by @mMerlin. The ioref label is in lowercase while the others are in capital case. Marks for the connectors could be created. 

Two Arduino uno parts: the one created by @mmerlin at the left, the one in the core at the right. See the long connectors in the core part and that the connectors are not highlighted in red as the resistor. 
[image]

So, what is the policy for the sch view parts? If we agree on it, we could create new parts with the new policy and slowly start to modify the old parts to the new policy.

vanepp, 2023-08-06:
OK there is a part for this board available here:

https://forum.fritzing.org/t/arduino-uno-r4-wifi/20145

This one has my usual (matches breadboard) schematic as we haven't yet decided to do something different, and so may change in future (as we look to be leaning towards abstract schematics.) As noted I'm not sure pcb is accurate.

failiz, 2023-08-07:
Seems good. A few suggestions for improvement:

* The color in the BB view seems very dark. In their 3d viewer seems more blueish:
[image]

* I think it is compatible with the Arduino Uno R3. So, the connector placement should be the same. You could also the interactive viewer where you can view and measure the PCB:  Check at the end of this page in the resources section (interactive viewer and select the PCB): https://docs.arduino.cc/hardware/uno-r4-wifi

* In the SCH view (apart from the fact that is not organized logically ;-) ), the I2C text is a bit far away from the line that groups the I2C pins. That line is also too close to the A5 pin, see below.
[image]

 * And being picky, I would use the same length for the horizontal lines that group the connectors on the right (like in the group at the bottom, that is short horizontal lines, maybe 0.05inches to set it to some number that we can reproduce later).
[image]

* In the PCB view, not sure if we need to include the ESPI and ESP32 connectors.

failiz, 2023-08-07:
I just realized, a few other things:
* The family is set to "Arduino Uno R4 Wifi". This breaks the homogeneity with previous arduino boards that had family="microcontroller board (arduino)"
* The variant is "variant1" which is unnecessary as there are no more variants for now. 
* There is another property, "editable pin labels", which seems like a bug

My recommendation is to use the same family used in previous parts "microcontroller board (arduino)". The type could be set to "Arduino Uno" and the variant to "R4 WiFi". This will need to change the variant and type of the other Arduino Uno boards. If not, just set the type field to "Arduino Uno R4 WiFi".
If there are more variants of the boad (e.g., without the I2C, ESP32 and ESPI connectors), we should add something that indicates that this part has the I2C, ESP32 and ESPI connectors. 

See the current menu of the type of an Arduino board:
[image]

vanepp, 2023-08-12:
In addition if we are trying to match the other Arduinos all my new ones need to change from layerId breadboard (current) to breadboardbreadboard (used by the standard arduinos.)

jiapei100, 2023-12-23:
[image]

This is what I get... So dark...

jiapei100, 2023-12-23:
Just use [https://forum.fritzing.org/t/arduino-uno-r4-wifi/20145](https://forum.fritzing.org/t/arduino-uno-r4-wifi/20145)

solved my problem. Thank you ...

KjellMorgenstern, 2023-12-23:
@jiapei100 This issue is about verifying the part and adding it to Fritzings library

DIYables, 2026-10-29:
The fritzing part for Arduino Uno R4 WiFi is available here https://forum.fritzing.org/t/arduino-uno-r4-wifi-fritzing-part/34723

[image]

KjellMorgenstern, 2026-10-29:
Any reason you are using Arial? It is a propietary font from Microsoft, so it doesn't work well in Fritzing.

Parts should use Noto Sans or OCR-Fritzing-mono, both are shipped with Fritzing.

DIYables, 2026-10-30:
> 23\. Any reason you are using Arial? It is a propietary font from Microsoft, so it doesn't work well in Fritzing.
>      Parts should use Noto Sans or OCR-Fritzing-mono, both are shipped with Fritzing.

Dear @KjellMorgenstern 

Arial front looks better on my PC. I will apply your suggestion font for next components, and will correct existing board later.

Thank you