fritzing/fritzing-parts #379: Added Pin Labels to MicroSD Card module

Tombstone2K, 2023-07-20:
I edited the included part for the microSD card module, adding labels for the various pins.
I have attached an image showing the original part and my edited part as well as the fzpz file (inside the attached zip file).

[image]

[microsdmodule_label.zip](https://github.com/fritzing/fritzing-parts/files/12112460/microsdmodule_label.zip)

How do I help merge these changes ?

KjellMorgenstern, 2023-07-22:
To help further, you can create a pull request, we update the modified SVG file (the breadboard svg in this case). 

For this simple change it can be done via the github webfrontend.

KjellMorgenstern, 2023-07-22:
"create a branch" to get started
[image]

Tombstone2K, 2023-07-23:
Thanks for your help.

I am not able to find a "create a branch" option.

[image]

It seems like I need to be given some permissions before I can create a branch.

Regards

KjellMorgenstern, 2023-07-23:
No need to be in the core team to make a PR. Please check with a github tutorial, it is essential knowledge.

Tombstone2K, 2023-07-23:
> No need to be in the core team to make a PR. Please check with a github tutorial, it is essential knowledge.

I created a Fork, committed the change to that, and opened a pull request in the main repo.
Thanks for your help

KjellMorgenstern, 2023-08-17:
I reviewed the part, and it works :-) 

But since we want to improve our parts format (and the workflow to add new parts), forgive me if I am a bit picky.

Which tool did you use to insert the text?
The text tags are quite convoluted:
```
<text
   xml:space="preserve"
   transform="matrix(0,0.75000031,-0.75000031,0,31.587174,-10.779515)"
   id="text1047"
   style="font-size:5.33333px;white-space:pre;shape-inside:url(#rect1049);fill:#f9f9f9"><tspan
     x="30.642578"
     y="31.739384"
     id="tspan621">CS</tspan></text>
```

So there is a text element, that has a transform matrix. Ok.
But: It also references to be  shape-inside "rect1049". This will offset quite a lot of tools.
Then, the text is inside a tspan, which again has x y coordinates. 

Fritzing renders it correct, but more or less by random: Fritzing simply throws away most style attributes, which are not valid TinySVG, so it ignores the "shape-inside" .

Fritzing tries to interpret the font-size, but the font-size is given in px, which is somewhat awkward if you take into account that the text transform matrix has some scaling, and the whole SVG is specified in inches in the header.
I think Fritzing patches this by ignoring the "px", and does not try to use the actual resulting pixel size. From a quick search: The relationship between px and physical units has been defined in CSS 2.1 . Fritzing uses TinySVG, which does not allow any CSS.

From the line formatting, it looks like the FritzingCheckParts script has been applied, is that correct? The script fixes some things, but sometimes makes it quite hard to review changes because of the reformatting.

Because of the header, I assume Adobe Illustrator was used, but typically illustrator does not add a width/height in inches. Did you fix that manually?

Tombstone2K, 2023-08-20:
@KjellMorgenstern 
I had used Inkscape to add the text labels to the pins.

I am a bit new to vector graphics. 
Thanks for fixing the issue if any.

Is there anything further required on my end ?

Regards