fritzing/fritzing-parts #397: Add scour to pipeline

KjellMorgenstern, 2024-09-05:
For nearly all recents parts I added or modified, I applied scour (https://github.com/scour-project/scour) to minimize, clean und unify the SVG files.

It would make sense to add this to our pipeline.

KjellMorgenstern, 2024-09-05:
Suggestion: 

Typically, we can't modify the files during the CI this would require an new commit.
We could run scour on each SVG with some sane settings.
If the file changes, then the check fails with an error message, with a link to some place that explains how to use scour in fritzing SVG files.

mMerlin, 2024-09-05:
Perhaps it can be added to 'pre-commit' instead. I think that allows content changes.  CircuitPython has that on their libraries, to enforce some linting.

I don't know about additional runs with the same parameters making additional changes. That could be a problem.